betaflight/betaflight #143: Motors jittering

brainmatterfpv, 2016-01-27:
Hi Boris, 

Sorry I have already flashed to 2.3.4 tonight and installed a new dodo FC and I'm still having the same issue as before. 

Please see below previous post

Hi,

I know its fairly normal for the motors to spool up with airmode enabled.. however mine are doing something very different (they weren't doing this with a previous version of betaflight)

This video (not mine) is exactly what mine are doing whenever I enable airmode.. causes my motors to get very hot as well.

https://www.youtube.com/watch?v=eZ37S8MTqGE

Need some help please!!

Thanks

borisbstyle, 2016-01-27:
try reflashing and resetting all settings. Also try another board as this is not experienced by anybody else.

The other user you are refering had Dterm jitter, which is not there on a healthy setup

brainmatterfpv, 2016-01-29:
Hi Boris, 

I've tried reflashing and resetting all settings, as well as using a different board entirely. 
I also swapped all my ESC's out for brand new ones tonight and it still happening. 

When you say Dterm jitter.. is this something that you might not notice using clean flight but becomes noticeable with betaflight and specially with airmode? Because I couldnt see any issues when I tested with cleanflight the other day. 

Any help would be appreciated!

borisbstyle, 2016-01-29:
It is being masked by non-Airmode actually as there are no low throttle corrections etc.

brainmatterfpv, 2016-01-29:
Do you have much of an idea what may cause it?
Only place I can think it comes from is my motors..

And is there anyway to fix it.. Or would I need to just not use betaflight?

Thanks for your replies

borisbstyle, 2016-01-29:
@brainmatterfpv 
I heared some rumors that some ESC's could cause some vibrations when damping light enabled what can affect some FC's. 
You can increase filtering and/or decrease D to even lower values.

set soft_gyro_lpf_hz = 40
set dterm_lpf_hz = 15

and see how it goes.

borisbstyle, 2016-01-29:
I will close this issue as it is not code related

closed 2016-01-29 by borisbstyle

jsweeney91, 2016-02-23:
This does appear to be a code issue. Same thing happens for me, I've had to switch back to CF until its fixed, tried adjusting most settings

borisbstyle, 2016-02-23:
lower yaw P

jsweeney91, 2016-02-23:
will do tonight, I'll get back to you

brainmatterfpv, 2016-02-25:
Tried this with two of my quads - jitters at first, lowered yaw P and it improved significantly.

I'll try to take them out for a proper flight to make sure it's all good and report back!

Thanks Boris