betaflight/betaflight #148: Lux FC Spektrum sat. issue

Rossbow, 2016-01-29:
I just finished setting up my Lux FC and I am using a Lemon sat. long antenna version, it works fine but as soon as I connect the board to the usb port the sat. turns off, and I can't see any control in the receiver tab. But I tried another brand short antenna version I had lying around and it work fine, It powers up on usb even without the main battery, It also powers up with the main battery and usb together. 
I have tried 3 Lemon sats. that I know work on my naze, but none will work with the Lux.
Any help would be appreciated

readerror67, 2016-01-29:
@Rossbow Do you have the 'SRX' and '3' jumpers soldered? Could you take a picture of how things are hooked up also.

readerror67, 2016-01-29:
BTW, you should probably move this over to http://www.rcgroups.com/forums/showthread.php?t=2554204 to keep the issues related to code things, just close this out and post the follow up on RCG.

closed 2016-01-29 by borisbstyle

Rossbow, 2016-01-29:
@readerror67  Yes SRX and 3 are jump soldered
I posted over there and got no response