betaflight/betaflight #151: powercube, drifting after a flip/roll

mavdnos, 2016-01-31:
i just came back to betaflight due to various reasons, but with the new airmode my quad drifts after a flipp or roll. Example: im flying 0 degrees tilt. i do a 340 degree flip, and after that i kinda drifts the remaining 20 degrees over 1-1.5 seconds to "level" again.

borisbstyle, 2016-02-01:
Are you talking about horizon mode here or level mode?

freddiebd, 2016-02-07:
I have the same problem in acro airmode

borisbstyle, 2016-02-07:
if you have drifting in acro mode you just have to tune the PIDs :)

Guys i dont see any software issues here.

Those who fly level just keep the default settings and dont change to 2 or
2.6khz.

If there is a drift in acro use PIDs to tune that out.

borisbstyle, 2016-02-08:
I am going to close this one as it is tuning related

closed 2016-02-08 by borisbstyle