betaflight/betaflight #152: Lux Target

Timbone1343, 2016-01-31:
When I use the Lux Firmware on my Lux I can no longer arm via Aux 1 even though the signal is in the arm location I setup on the Modes tab.
The save and reboot also does not initiate a reboot. I have to disconnect from CF and re-power the Lux before I continue to ensure a reboot.

Both of these features work on the Calibri Race firmware on the Lux so I'm using this for now.

I'm using:
Lux
Taranis x9d+
X4R-SB - sbus

readerror67, 2016-01-31:
So I am setting up one now about to go fly, I can't explain this since I haven't looked in to it much further yet, but when my mincommand was set to 1000, it was outputting a pulse of 125, it was too high for the settings I have on the LB20s, I dropped it to 980 and was able to arm.

Again, not sure if its related, just something odd I noticed and maybe worth trying on yours too.

djtaotao, 2016-01-31:
I had the same problem with Naze32 full rev5.
Here is the same issue: https://github.com/borisbstyle/betaflight/issues/94.

I solved by disabling baro and mag:
baro_hardware = 1
set mag_hardware = 1

jsweeney91, 2016-02-02:
try: 
set emf_avoidance = ON 
same ticket as @djtaotao referenced. I had the exact same issue on naze32 rev6. And as @readerror67 explained make sure your min_command values are more than your minimum throttle values

readerror67, 2016-02-02:
emf_avoidance wont work on this, it needs to be at the 72MHz since it uses USB.

jsweeney91, 2016-02-02:
can you clarify? I'm using USB and emf_avoidance worked for me

borisbstyle, 2016-02-02:
f3 targets dont support emf_avoidance.

closed 2016-02-08 by borisbstyle