betaflight/betaflight #153: Rates/Expo gone when activating airmode in BF2.3.3 at 2khz

konsumkid, 2016-01-31:
Hi,

When flying and activating airmode on SPF3 all my 3 quads show up an strange behavior. In acro everything is nice and sharp, but when I toggle the airmode switch - the copter goes very slow at movements. Maybe this is only an setting, but I ve used a clean install.. 

Many thanks for the great project.. Keep going!

borisbstyle, 2016-01-31:
I guess you have acro_plus_factor activated.

Disable acro_plus_factor and your rates will be normal

konsumkid, 2016-01-31:
Thx I will test it asap .. 

// Alexander Oelling

borisbstyle, 2016-02-01:
Yes acro_plus is only active in airmode. When setting that you automatically overrule the normal rates with acro_plus rates.

Either increase acro plus factor or disable it

closed 2016-02-01 by borisbstyle