betaflight/betaflight #162: AlienFlight F3

MJ666, 2016-02-06:
Please can we add the AlienFlight F3 build for future releases. More and more of this boards will be seen in the wild. Thanks.

borisbstyle, 2016-02-06:
you added that one right?

MJ666, 2016-02-06:
I meant the firmware hex on the release page. I have not added an hex in my PR. You can build it together with the other targets firmware in the future and put on the release page.

borisbstyle, 2016-02-06:
that gets build automatically from fake_travis if its there

MJ666, 2016-02-06:
OK, I have updated the fake_travis_build.sh and .travis.yml as part of my PR. Anyhow I have modified ALIENWIIF(X) to ALIENFLIGHTF(X) but in the past also the ALIENWIIF(X) targets where not part of the release builds for BetaFlight? I also can see AFROMINI but tit is not in the travis files but on you release page. So I was wondering.

borisbstyle, 2016-02-06:
The targets which were not included are the "untested" ones. When I get some feedback that something works I simply just add it.

Afromini was a special one, which I did separately. I just recently switched to fake_travis. I had a seperate script for compiling selected targets.

MJ666, 2016-02-06:
Cool then we should be set for the next release. All AlienFlight (F1, F3 V1, F3 V2) targets are flight tested last days. Thanks.

borisbstyle, 2016-02-06:
Gyro sync works?

borisbstyle, 2016-02-06:
I am still working on 2.4.0. Just one more feature to add and test. Maybe a few cleanups and it will get released

MJ666, 2016-02-06:
Yes gyro sync is working for most of the F3 versions. All F1 and some early F3 prototypes don't have the interrupt connected.

readerror67, 2016-02-06:
Does anyone actually have these boards??

borisbstyle, 2016-02-06:
I also never heared anybody complaining about it

MJ666, 2016-02-06:
The community is starting to build boards since a while. We already have released the designs to the public last year and some people started to build them. After some issues we just released the new GitHub repo with the updated designs last night. 

https://github.com/AlienFlight/Flight-Controllers

There are also some users wich are starting to build larger batches of this FC's.

readerror67, 2016-02-06:
Ah cool, glad to see they are open source. Remember hearing about the F3 one a ways back but not seen it come about but atleast people can make their own now.

borisbstyle, 2016-02-08:
added in 2.4.0

closed 2016-02-08 by borisbstyle

MJ666, 2016-02-08:
cool thanks.