betaflight/betaflight #165: Change 'Rateprofile' settings to allow for 'Profile' setting

mikeller, 2016-02-08:
Hi @borisbstyle.

First of all thank you very much for the clean up of the configuration, I think this has improved it a lot.

One small thing I wish I had back: With the old configuration, it was possible to set up a 3 position switch to allow 'Rateprofile' setting. In 2.4.0-RC1, this can still be configured, but the switch has no effect. Would it be possible to change this setting so that it allows switching of profiles by switch, so that it can be done in-flight, like it was possible with rateprofiles?

Another minor inconsistency in 2.4.0-RC1 is that in a cli dump, 'aux', 'adjrange', 'rxrange', and 'servo' settings are listed per profile, whereas as far as I can see they are stored only once in master, and not individually for each profile.

KiteAnton, 2016-02-08:
Is there need to switch rateprofile "in the air" or can't the normal switch profile with stick commands be used? 
Rateprofile was previously a separate config and now it is merged in the profile. When switching profile it writes current setup to eeprom and reads new settings, I am not sure if this is safe to do in flight? I commented this section out but it can be uncommented easily again if it is needed and "safe" .. ?

mikeller, 2016-02-08:
Hi @KiteAnton.

It is not so much that there is a need, more that this was a useful feature.
If it is not possible on a safe way any more then we'll have to live with it.
In this case it would make sense to entirely remove the option from the settings menu.

borisbstyle, 2016-02-08:
@mikeller 
That is IMO not a very good idea to change profiles as there are some other parameters in there besides rates.
To make that work some additional code is needed. Maybe something for the future, but I don't think it will be added as it may be confusing.

closed 2016-02-08 by borisbstyle

mikeller, 2016-02-08:
Hi @borisbstyle.

Fair enough. In this case 'Rateprofile' needs to be removed from the settings list, as it does not work any more, which is confusing.

borisbstyle, 2016-02-08:
That is a configurator support. Betaflight uses cleanflight configurator so
that cannot be fixed.

mikeller, 2016-02-09:
Hi @borisbstyle.

> That is a configurator support. Betaflight uses cleanflight configurator so

Not really, it seems to be persisted on the FC as well:

```
# dump profile

# profile
profile 0

[...]

# adjrange
adjrange 0 0 1 900 2100 12 1
```

KiteAnton, 2016-02-09:
Yes, that part can be removed. It is these lines that should be removed. 
https://github.com/borisbstyle/betaflight/blob/betaflight/src/main/io/serial_cli.c#L1809-L1819

The part that is included in the configurator is the ability to select that function in the adjustments tab. It is kind of hardcoded in the configurator what type of functions exists depending of interface version. I don't know why it is like that. 
https://github.com/cleanflight/cleanflight-configurator/blob/development/tabs/adjustments.js#L68-L76

etracer65, 2016-02-18:
I strongly want to second the request to restore the rateprofile capability.  Being able to change the rates while in flight is extremely useful.  For example I use a three position switch to have low, medium and high rates depending on the flying I'm doing.  Doing flips and rolls, use high rates.  Racing through gates, medium rates. Proximity, low rates.

I realize that rateprofile and profiles were confusingly mixed, but by eliminating rateprofile and the ability to switch on the fly you've removed an extremely useful feature.

mikeller, 2016-02-18:
In order to not revert back to the old and confusing way that profiles and rateprofiles were linked, how about making rateprofiles optional:

Keep the rate values stored in the profiles just as they are now. Add a section that allows to store 3 sets of replacement rateprofiles like

```
rateprofile 0 <rc_rate0> <rc_expo0> <rc_yaw_expo0> <thr_mid0> ...
rateprofile 1 <rc_rate1> <rc_expo1> <rc_yaw_expo1> <thr_mid1> ...
rateprofile 2 <rc_rate2> <rc_expo2> <rc_yaw_expo2> <thr_mid2> ...
```

Then, when the `adjrange` for rateprofile selection is activated, overwrite the rates with the ones stored in the `rateprofile` selected by a 3 position switch, independent of the current profile selection.

Your opinion, guys?

etracer65, 2016-02-18:
> In order to not revert back to the old and confusing way that profiles and rateprofiles were linked, how about making rateprofiles optional:

>Keep the rate values stored in the profiles just as they are now. Add a section that allows to store 3 sets of replacement rateprofiles like

> rateprofile 0 \<rc_rate0> \<rc_expo0> \<rc_yaw_expo0> \<thr_mid0> ...
> rateprofile 1 \<rc_rate1> \<rc_expo1> \<rc_yaw_expo1> \<thr_mid1> ...
> rateprofile 2 \<rc_rate2> \<rc_expo2> \<rc_yaw_expo2> [...]

Rather than change the syntax and maintain compatibility and consistency, I would recommend that the rateprofile command simply be "fixed" as a standalone feature.  Make it a mini profile that only stores the rates (like originally intended).  rateprofile 0 selects and loads the rates.  Any changes to the rates affect the selected profile.  So each real profile would have its own set of 3 rate profiles.

voyn238, 2016-02-18:
I'm still on 2.3.5 mainly because I don't want to lose functional rateprofiles. I like having mild/medium/wild rates available on a switch, and I'm really not a big fan of seeing it go away. I suppose I can create a similar effect on the radio itself, maybe adjusting between 60%, 80%, and 100% on pitch/roll/yaw, but that might cause problems since on anything less than 100% it will not register stick commands anymore. Maybe restoration of rateprofile isn't needed if some alternative can be discovered to change rates while in the air.

KiteAnton, 2016-02-18:
I guess, as you propose @etracer65, adding rateprofiles to profiles is one way to enhance this and it will be less confusing as it was previously were rateprofiles was a parallel structure. However, it will use additional flash + RAM with that addition. This could be investigated. 

Have you tried to use acro+ feature? That is possible to set up in modes tab and I guess it can replace some of the functionality you want. Enabling acro+ when doing flips/rolls and disabling it for proximity/race.

borisbstyle, 2016-02-18:
@etracer65
There is an acro plus feature on switch for more wild acro and has separate
parameter for rate speed. That feature is finally safe to use from RC9.

@KiteAnton
I recently did some massive cleanup by removing lowpass.c. I guess
rateprofile might fit.
But I am also adding some new stuff as well behind the scenes like PID3 so
yeah might be tight. But if you have some time please make a PR.
I guess it is just rc rate...rates and expo we need in there.

etracer65, 2016-02-18:
@borisbstyle 
Using acro+ might be an option for having normal and then extreme rates, but it's not very granular.  Also unless I'm misunderstanding there isn't any ability to have different rates for each axis when enabling acro+.  Also that acro+ doesn't affect yaw (maybe that's changed?).  In my case I prefer to have lower roll/pitch when I'm doing proximity but I turn the yaw up.

etracer65, 2016-02-18:
And regarding space - I'll echo what others have said about removing GPS, baro, etc.  If we're stopping things out like profiles, etc. to focus more on performance, why keep features that 99% of the betaflight users won't use?

etracer65, 2016-02-18:
And I agree that gtune can go as it hasn't worked in a long time.  Maybe it's time to make some of the less used features (like GPS for example) available only on F3 targets with more flash.  The downside of course is that this adds more fragmentation and confusion, but at least it gives a way forward.

KiteAnton, 2016-02-19:
So if adding this to profiles... is pitch/roll/yaw rate sufficient or is also rc rate and expo wanted "switchable"? 
I'm also a bit concerned that it might be confusing when using the gui what values actually is changed since there is no indication regarding active rateprofile. 
I might get some time during the weekend to elaborate this enough for a PR, stay tuned.

Also, could we "lock" rc rate to 1.0 (I.e. not make it changeable) to ease up configuration when only changing rate matters? (Maybe this is too radical change and not at all wanted..?)

etracer65, 2016-02-19:
I would definitely like to retain the expo settings as I do sometimes change them - particularly yaw_expo (I run really high yaw rates).  Ideally it would be best to restore all of the parameters that made up the previous rate profile:

set rc_rate = 100
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

I'm not sure about tpa_rate and tpa_breakpoint as those are not rates but PID tuning.  I personally wouldn't care about thr_mid and thr_expo as I don't use them (maybe the 3D guys?).

I also thought about the GUI and don't have any good ideas.  We may just have to wait until the GUI catches up.  Maybe the fixed rateprofile capability that's disconnected from the profiles will filter back not CF and drive the configurator.  With the reduction to 2 profiles it's already confusing in the GUI to a certain extent as you can still select profile 3 but of course it doesn't work.

I don't really have an opinion on rc_rate.  I always leave it at 1.0 so to me it doesn't matter if it's in the rateprofile.  Not sure if others will have a problem with that.  But if the goal of retaining all of the previous values from a compatibility and migration from CF is important then it becomes moot.

Do you think it will be possible to have 3 rate profiles like before so that it matches up with the selection ability under adjustments?

nutmegfpv, 2016-02-20:
I too admit that I was a big fan of the ability to change rates mid-flight.  I did not find it confusing to have profiles and rate profiles separated.  Three separate profiles and rate profiles weren't necessary, but two of each would be nice.  For now, I'll live with what is available.

mikeller, 2016-02-20:
>  Maybe the fixed rateprofile capability that's disconnected from the profiles will filter back not CF and
> drive the configurator. With the reduction to 2 profiles it's already confusing in the GUI to a certain
> extent as you can still select profile 3 but of course it doesn't work.

That's exactly what I meant in my initial proposal: Have selectable rateprofiles that are only activated when the 'Rateprofiles' setting is assigned to a 3 position switch and activated. Setting the selectable rates would not be supported in the configurator, but at least it would not interfere with the existing configurator settings, as none of them would be affected.

As for me, I would like to see all of the original rates part of the rateprofile, except for tpa_rate and tpa_breakpoint. But that's just my opinion.

KiteAnton, 2016-02-20:
I made some tests and got this working. Added rateprofiles as part of profiles. In some sense this is more logical but it will use up more memory compared to how it was before with a parallel structure. What is your opinion regarding this? I can clean it up and make a PR soon.
Unfortunately the data and code consumes an additional 544 bytes flash and 60 bytes RAM. 

I have tested playing around with parameters, tried switching through radio etc. but unfortunately it is snowing here now so I will not be able to test this flying.

mikeller, 2016-02-20:
Hi @KiteAnton.

Thanks for your effort, having the rateprofiles back is nice.

One small thing:

The rates are still reported as part of `dump profile` in CLI:

```dump profile
[...]
set rc_rate = 50
set rc_expo = 50
set rc_yaw_expo = 40
set thr_mid = 55
set thr_expo = 70
set roll_rate = 0
set pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500```

This can lead to confusion / faulty configuration after dump / restore, as the settings for the currently active rateprofile are overwritten when restoring, and not the one for the profile that was active when `dump profile` was executed.

mikeller, 2016-02-21:
Sent a pull request that fixes this.