betaflight/betaflight #166: 2.4.0-RC3 = where is "failsafe_procedure" (CLI) ?

TiTidom-RC, 2016-02-08:
Hi Boris,

I just flashed the last (2.4.0-RC3) BTFL firmware on the LUX, i see the new failsafe TAB with all options, but in the CLI, i can't retrieve the "failsafe_procedure" var ? it doesn't appear either in a dump.

Did i make any mistake ? or the var is missing (in Cleanflight 1.12-RC2, i can get this var in the CLI)

Thank's,

borisbstyle, 2016-02-08:
@TiTidom-RC 
Not supported yet.

borisbstyle, 2016-02-08:
its hardcoded...cant change it in GUI btw. It will reset it for you

borisbstyle, 2016-02-09:
added in RC6

closed 2016-02-09 by borisbstyle

TiTidom-RC, 2016-02-09:
Perfect, thank's a lot ! ;)

TiTidom-RC, 2016-02-09:
@borisbstyle : i just tested the RC6 : the failsafe_procedure is there, but in CLI, possible values are ON and OFF. In Cleanflight it's 0 and 1. is that a choice to use on / off instead of 1 and 0 ?

Thank's.

PS : on this RC6, on the LUX FC, after each "save and reboot", i have to click on disconnect and connect again, as if the reboot "trigger" is not detected anymore ?