betaflight/betaflight #167: Request Cleanflight Jeti EX Bus Protocol Radio Merge from Cleanflgiht pull #1709

SnipeXR, 2016-02-08:
Like to request to have this clean flight pull request get merged with Betaflight
This brings in Jeti Duplex EX Bus support and telemetry. 
https://github.com/cleanflight/cleanflight/pull/1709/files

borisbstyle, 2016-02-08:
Ok got it here.....will merge after short review. Thanks!
https://github.com/borisbstyle/betaflight/pull/168

borisbstyle, 2016-02-08:
Added

closed 2016-02-08 by borisbstyle

SnipeXR, 2016-02-08:
Thank you @borisbstyle

paulboes73, 2016-07-03:
Has this been merged?  I don't see it with 2.9.0 or 2.8.1 of betaflight.  Please let me know, because I'm using a Jeti DS-16 and would like to have Telemetry working?

SnipeXR, 2016-07-03:
Yes Jet EX Bus with telemetry has been interested since 2.7.x. 
I have used it and works great. 
The beta flight configurator is missing the option to select ex-bus you have to do it via CLI set serial_rx = Jeti-EXbus

paulboes73, 2016-07-04:
Thank you SnipeXR for the quick respond back.  The receiver is working on JETIEXBUS.  I noticed the command to be a little different. I had to use  "set serialrx_provider=JETIEXBUS" is the command on used to get the receiver working.

paulboes73, 2016-07-04:
How did you get the Telemetry to work?

paulboes73, 2016-07-14:
I'm having trouble on the Telemetry piece of this configuration.  My serialrx_provider=JETIEXBUS works fine and I see that in the receiver tab.  However, on the ports under telemetry I don't see anything for JETI, only HOTT, SmartPort, FrySky, and LTM.  I have the 2.7k 1.8w power resistor connected to a UART port on the TX out to the receiver signal.  What I'm I missing on this setup,  is there another command line for JETI telemetry?

paulboes73, 2016-07-14:
Okay, here is an update.  Apparently, this will not work with my Duplex R6 light receiver.  However, I connected my R3/RSW receiver and everything works great.  Thanks for the quick feedback mikeller I thought the same thing when reviewing the 1709 that I didn't need to set something else to get it working.  I was just getting frustrated with my R6 receiver and thought it was something else in betaflight that I missed in the setup.  Everything is working now with the R3/RSW receiver.

Benjamin000, 2016-12-19:
Hi,
I see you guys have be able to connect your jeti receiver using betafligh 
I'm new to this and I would like to ask for your guy's help on how to set the receiver as choosing the RX type for the  jetiexbus Im using the duplex R3 receiver
any help will be much appreciate