betaflight/betaflight #172: Lux crash

gruntwerkz, 2016-02-09:
I installed the new 2.4 on my lux FC, the firmware flash took and I was able to connect once. After I loaded my settings and did a save/reboot it lost communication with clean flight. Then I received an error that windows no longer recognized the device, tried re-installing the STM32 drivers and rebooted PC, no luck. Had to perform the rescue function and install the previous version of beta flight, something isn't right with the lux file.

travmo, 2016-02-09:
Works fine for me, i'm currently using the latest 2.4.0 RC 6 but it worked fine RC4 and RC 5 as well. This might be a better comment for the RCG forum page. Link: http://www.rcgroups.com/forums/showthread.php?t=2464844

borisbstyle, 2016-02-09:
Did you try disabling LED_STRIP?

gruntwerkz, 2016-02-09:
Hey guys, I was told it's because I tried to load my settings from the previous 2.3 firmware. I will test that out in just a minute, I would like to keep LED on.

gruntwerkz, 2016-02-09:
So I installed the firmware again, same thing happened. Did not load any new settings, only tried to save the settings I changed. My PC longer recognizes it. I will have to recover it again for it to connect.

borisbstyle, 2016-02-09:
You need to get red of ledstrip feature somehow. Try flashing cf 1.12 first

borisbstyle, 2016-02-10:
SOlved in RC7

closed 2016-02-10 by borisbstyle