betaflight/betaflight #173: Support for CJMCU board

isilcala, 2016-02-09:
I noticed the support for CJMCU board was dropped due to it's '64k rom' but it actually has 128k rom which is enough for the hex. Please add it back.

borisbstyle, 2016-02-10:
Sorry I have no feedback if that board works.

contactsimonwilson, 2016-02-17:
I can confirm it works. Currently running the 2.1.0 release that I built for that target locally. I can build against the latest release and verify it works if needed.

isilcala, 2016-02-19:
it woks with cleanflight fine. should be able to work with betaflight as well?

borisbstyle, 2016-02-19:
does the gyro sync work.....do you get solid looptimes like exactly 1000us
on default settings or is it 1100us?

borisbstyle, 2016-02-23:
I will close this issue as CJMCU is not going to be supported for now. Lack of support / testing etc...

closed 2016-02-23 by borisbstyle

isilcala, 2016-02-24:
with a default cleanflight 1.12, I can get looptimes between 990 ~ 1000us.