betaflight/betaflight #174: SPRacingF3Mini : 2.4.0-RC6 - FC Bricked on serialRX activation (recovery via boot pad OK)

TiTidom-RC, 2016-02-09:
Hi Boris, 

While testing the 2.4.0-RC6 on the F3Mini, my FC was bricked (red led still fix and impossible to connect, the USB port was not recognized anymore).

I searched and found a way to reproduce that : 

- flash betaflight rc6 on F3Mini with "full erase chip"
- connect to the FC
- "Ports" tab : uncheck MSP on UART1 + save and reboot
- go to "configuration" tab
- click RX_SERIAL + SBUS + "Save and reboot"
- after reboot, go to ports, and activate "Serial RX" on UART1 + "Save and reboot"
- on reboot, FC is bricked, red led still ON and usb port not recognized anymore

PS : if "serial rx" on ports tab is activated BEFORE going to configuration to click serial + SBUS, it's working.

If i use boot pads, i'm able to flash firmware again.

with CF 1.12.0-RC2, it's working, no bricked FC with the same test.

Thank's,

PS : sorry for my english...

borisbstyle, 2016-02-09:
thanks for reporting.....I will test this soon.

borisbstyle, 2016-02-10:
I reproduced this.

I will park this as it doesnt seem that crucial.

hydra, 2016-02-18:
@TiTidom-RC it's not bricked if you can recover from it.  The term bricked implies you can't use the device again. :smile:

TiTidom-RC, 2016-02-20:
@hydra : hummm... yeah you are right... but to be honest, i didn't know how to describe this state with other words (in english...) :smile: : blocked perhaps ? :laughing:

hydra, 2016-02-20:
@TiTidom-RC `unresponsive`

devinpearson, 2016-03-03:
I had similar issues yesterday with F3mini. not sure which operating system @TiTidom-RC used but it kept becoming unresponsive on my mac. If you hold down the two buttons for 5 seconds when you apply power it connects again. 

Ive tired reproducing the problem with a windows machine and havent been able too

bevibes, 2016-03-03:
Same issue for me... Freeze of the spracingf3 mini when I select serial rx on uart1...

closed 2016-07-03 by borisbstyle