betaflight/betaflight #177: Support for KISS FC

gregorg, 2016-02-10:
Kiss flight controller from flyduino is compatible with SPF3 boards, it used a STMF3 processor. Maybe there is nothing to do to make it compatible ...

Even if Kiss firmware is good, it will be great to compare with Betaflight. And betaflight offers many tools like testing one motor per one...

borisbstyle, 2016-02-10:
Work in progress...
https://github.com/borisbstyle/betaflight/tree/betaflight_kiss

gregorg, 2016-02-24:
Hi,

I've compiled target KISSFC, convert .hex file to dfu, uploaded it to FC and ... the blue led flashs then the FC enter bootloader mode.

What can I do for help ?

borisbstyle, 2016-02-24:
that code is not finished..... a lot of things to do to make it work.

barthalamu, 2016-05-10:
Has any more progress been made to this? There are quite a few of us on the KISS FC RCGroups forum that are having major issues with the KISS firmware. I'd love to switch to Betaflight without having to swap out the FC.

Thanks!

dgddustin, 2016-06-21:
when will the kiss fc finally be supported? the kiss firmware sucks balls right now and i absolutely love betaflight and would love to run it on my kiss fc instead of switching fc's

tannewt, 2016-06-21:
The target is checked into the development branch but probably still needs testing.

deepimpactio, 2016-06-21:
Tell us how you really feel lol

dgddustin, 2016-06-21:
Ill be a tester lol just shoot me the firmware! I dont really kno how to use this site

martinbudden, 2016-06-21:
The plan is to have a KISSFC target as part of the 3.0 release. There is a KISSFC target in the development branch - this supersedes the version in the betaflight_kiss branch. It builds but is totally untested - I don't have a KISS FC. Anyone is welcome to build this and report back any problems.

dgddustin, 2016-06-21:
could you build it and send me a link? like i said i do not know how to use this site all that well or i would do it and test.

AndersHoglund, 2016-06-21:
Everything commited to the betaflight development branch is built here: [http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Development/](url) . Please feel free to download the build artifacts zip file.

dgddustin, 2016-06-21:
when clicking on the like it says page not found

AndersHoglund, 2016-06-21:
Try again, some strange handling of links. Or copy & paste this URL http://andwho.sytes.net:8080/job/BorisB_BetaFlight_Development/

dgddustin, 2016-06-21:
i had to copy and paste thanks, also iv successfully downloaded firmware files and i created a dfu file and loaded it onto the kiss fc and guess what it works!!!!!!!!! i am now working on changing the motor order from Kiss's motor order to betaflights order then i will test flight and report back thanks

martinbudden, 2016-06-21:
I think you can just set the mixer to "QUADX1234". Not sure if you can do this in the configurator, but I'm pretty sure you can do this in the CLI.

dgddustin, 2016-06-21:
to late now haha its done and flying i just need to add alittle  more filtering

dgddustin, 2016-06-22:
Ok so I now have it flying but am having issues with I term and d term. I have major dropping in pitch and roll when punching out and I have I term at like .055 for both pitch and roll. Also I having issues with lots of bounce back after flips and roils and I have d term at almost 60 for both pitch and roll. That seems like a lot with no fix to the issue. I'm using luxfloat

borisbstyle, 2016-06-22:
I think the dev branch first needs to be rebased/merged with the 2.8.1 to
get better tuning, but also other things need to be verified on this target
as well.

Thanks for testing. Are you using SBUS by the way?

dgddustin, 2016-06-22:
blockquote, div.yahoo_quoted { margin-left: 0 !important; border-left:1px #715FFA solid !important; padding-left:1ex !important; background-color:white !important; }  Okay let me know when or if some things change so I can test them. 🤓
Yes I'm using Frsky xsr with sbus 

> Ok so I now have it flying but am having issues with I term and d term. I
> have major dropping in pitch and roll when punching out and I have I term
> at like .055 for both pitch and roll. Also I having issues with lots of
> bounce back after flips and roils and I have d term at almost 60 for both
> pitch and roll. That seems like a lot with no fix to the issue. I'm using
> luxfloat
>
> <Support for KISS FC · Issue #177 · betaflight/betaflight 
  
|  
|   
|   
|   |    |

   |

  |
|  
|   |  
Support for KISS FC · Issue #177 · betaflight/betaflight
 
Kiss flight controller from flyduino is compatible with SPF3 boards, it used a STMF3 processor. Maybe there is nothing to do to make it compatible ... Even if Kiss firmware is good, it will be gre...
  |   |

  |

  |

 

> <Support for KISS FC · Issue #177 · betaflight/betaflight 
  
|  
|   
|   
|   |    |

   |

  |
|  
|    |  
Support for KISS FC · Issue #177 · betaflight/betaflight
 
Kiss flight controller from flyduino is compatible with SPF3 boards, it used a STMF3 processor. Maybe there is nothing to do to make it compatible ... Even if Kiss firmware is good, it will be gre...
  |   |

  |

  |

 
let

tannewt, 2016-06-22:
Weren't they already merged?
https://github.com/betaflight/betaflight/commit/897aab9c86609aa6497675ede77ec21de73cd1a3

dgddustin, 2016-06-24:
has there been any updates to the kiss target?

llambkin, 2016-06-25:
I have a weird issue where I loaded mixer quadx1234, and now I cant do anything. I have no copter in the setup page, and I cant click on anything else. I will reflash and see if I can reproduce.

llambkin, 2016-06-25:
OK, I puttied into the quad and changed the mixer back to quadx, and the gui now responds. I tried remapping the motor outputs using a custom motor mix, and it doesnt appear to work. Also VBAT doesnt work. I think I might wait for more work to go on before I try again.

llambkin, 2016-06-25:
Alright! I now got the mmix sorted. For anyone else trying to get it going, and quadx1234 doesnt work for you, here is the mmix numbers required to get it off the ground:

mixer CUSTOM
mmix reset
mmix 0 1.000 1.000 -1.000 -1.000
mmix 1 1.000 -1.000 -1.000 1.000
mmix 2 1.000 -1.000 1.000 -1.000
mmix 3 1.000 1.000 1.000 1.000
smix reset

llambkin, 2016-06-26:
Well, all good on my end. I have flown about 12 packs through it so far, and its going good. All I need now is Telemetry and Vbat and I am a happy man. 
Sample footage and blackbox working :D
https://www.youtube.com/watch?v=bhA3OeGlq-E

gregorg, 2016-06-27:
is ESC passthrough available and working ?

llambkin, 2016-06-27:
Hmmmmm, havent tried that one. I will stick it on my other KISS beast and give it a try. I am doubtful, the KISS FC uses 5v PWM drivers, so maybe not/ever. I will report back here.

llambkin, 2016-06-27:
Thats a big negative on the ESC passthrough. It uses a virtual com port, and 5v PWM drivers, I am doubtful that we will get passthrough support with the KISS FC.... :(

llambkin, 2016-06-27:
I did more testing this evening. The RX3 TX3 pads on the bottom of the board line up to uart3 in the Betaflight Configurator. So all UARTs are accounted for. UART 1 is on the top corner simply labelled as RX TX, and UART 2 is in 2 places, one being the PPM/UART pads and the other on the underside of the board.

Id like to know how they got the extra inputs for individual telemetry from ESC's and weather this information can be imported into Betaflight.

The only thing holding this build back is the lack of Vbat monitoring, current would be nice, but I doubt it would be an easy task

llambkin, 2016-06-27:
I have done some probing on the voltage dividers that come off the battery input. It pins out to pin30 (PB11) on the F3 chip. I would make the changes myself, but I dont know how to do it.

martinbudden, 2016-06-27:
@llambkin , are you sure it's PB11 - the preliminary code for the KISS FC says it is on PA4.

Also, I've just looked, and UART3 RX is on PB11.

llambkin, 2016-06-27:
I noticed after I had a poke about the code. There is a divider on the board that pins out to there, however I currently have my OSD working on UART3, so the initial code is correct for that. 
The voltage divider in question is on the under side of the board, right next to the + input and it pins out to the 3rd pin in from the bottom right of the chip.

martinbudden, 2016-06-27:
I don't have a KISS FC, so I'm afraid I can't check board traces.

llambkin, 2016-06-28:
I do have a spare board outside of a craft, I will have a further probe
into it tonight and see if I can try my hand at compiling something that
works, at the very least I will identify the pins and report back here.

llambkin, 2016-06-28:
OK. I was reading the chip wrong.
Apparently it is on PA0
Here is an image of where the dividers go to.
https://drive.google.com/open?id=0B1_e81Muw9i6YnQ5RUphOGpJVVE

And looking at the pinout reading of an F303. That is PA0. I might have a go at coding and see if I can make the changes myself by doing a git pull, and see if I can get my head around compiling into a hex.... No better time to learn :D

mikeller, 2016-06-28:
@llambkin:

If you are struggling with building the firmware on your setup, this might help:

https://hub.docker.com/r/mikeller/betaflight-build/

llambkin, 2016-06-28:
Well.... I cracked it..... It is on ADC1 PA0.
How do I make the change to be submitted to the code?
Or can someone with rights do this?

martinbudden, 2016-06-28:
@llambkin , I'll submit a change this evening.

llambkin, 2016-06-28:
Cool, do you know if we can use VCP passthrough programming for ESC's? it is using a similar setup to the SPRF3, but I believe that it has 5v PWM drivers for the motor outputs. Do you know if this will pose a problem or not?

llambkin, 2016-06-28:
I have gone through the motor order page (in target.c) and redefined the motor outputs to match up with how people have their KISS crafts set up (also matches the PWM# on the motor output pads)

The motor order is:
    PWM7  | (MAP_TO_PPM_INPUT << 8),
    PWM3  | (MAP_TO_MOTOR_OUTPUT << 8),
    PWM2  | (MAP_TO_MOTOR_OUTPUT << 8),
    PWM4  | (MAP_TO_MOTOR_OUTPUT << 8),
    PWM1  | (MAP_TO_MOTOR_OUTPUT << 8),
This way I don't have to use a custom mixer.

llambkin, 2016-06-28:
Can someone please look into this:
http://ultraesc.de/downloads/KISS_telemetry_protocol.pdf
And see if we can implement it? Even if we can use it as a source of combined amperage (instead of having to have a separate current sensor, say have a checkbox "KISS ESC Telemetry"

I will see if I can find out where on the board the Telemetry comes in on the CPU

We may even be able to use it with other FC's

martinbudden, 2016-06-28:
@llambkin , can you raise the KISS ESC telemetry protocol as a separate issue - it is to do with KISS ESCs, and as you point out, could be used with other FCs.

martinbudden, 2016-06-28:
@llambkin , I've updated the KISS target to use VBAT as you specified and also fixed the motor order.

Can you retest this and confirm it all works OK?

As for using VCP passthrough for programming the ESCs, I'm not a hardware guy, so I'm not really sure. As I understand it the KISS FC uses FETs to buffer the PWM output and this probably precludes the use of serial passthrough, but I think it also depends on the type of FETs and how they are wired up, so it may be possible.

llambkin, 2016-06-29:
So..... I updated and still get this.
[image]

martinbudden, 2016-06-29:
@llambkin , the flag for serial passthrough is not set for the KISS FC, so of course it won't work.

As I said, it may be possible to get it working, but it needs someone who understands how the FETs are connected on the KISS FC to do so.

If you want, you can try adding the `#define USE_SERIAL_4WAY_BLHELI_INTERFACE` flag to the KISS `target.h` file and rebuilding, but I very much doubt that will work.

llambkin, 2016-06-29:
Okie dokie, I did that, and it still doesnt work. So I guess I will leave it at that. I also thought I'd mention that the Vbat is not enabled by default and the stock scale is incorrect. I have changed mine to 164 to make it accurate. Can this be defined to make it default?

martinbudden, 2016-06-29:
@llambkin , I've defined the default VBAT scale to be 164, and set the feature on for KISS. I've also defined the default receiver as PPM.

Can you rebuild and check this is all OK?

llambkin, 2016-06-29:
Jenkins beat me to the build :D 
I can confirm that VBAT is working and to the correct scale. The default of PPM is not working. I think this is because KISS FC switched the serial port UART2 into a PPM input depending on which RX was selected. I dont know if Beta/Cleanflight is capable of doing this. All of my other crafts use RX serial, but my test bed has a D4R-ii running PPM

martinbudden, 2016-06-29:
@llambkin , what do you suggest should be the default RX setting?

llambkin, 2016-06-30:
For default usage, I believe that we should use RX serial (SBus). Not only
because most use FRsky stuff, but also because its attached directly to
UART2.

I will do further investigation into the pads next door to the RX input and
see if we can use that with PPM, then we can put instructions specific to
KISS into the wiki showing which pads do which function. I'd like to find
the softserial pads also.

martinbudden, 2016-06-30:
@llambkin , I've updated the KISS target to use default to RX serial, SBus. Can you check this is OK?

gregorg, 2016-07-02:
@martinbudden : default SBUS is OK, however default RX serial not. I have to configure port UART2 to Serial RX and then RX serial in receiver mode to get it working.

gregorg, 2016-07-02:
Some more testing:

- failsafe is OK
- voltage scale is more accurate at 160
- VFAS is not transmitted through the telemetry
- buzzer is not working

llambkin, 2016-07-02:
I can confirm with the Buzzer, it pins out to PB13 on the CPU as per the target.h. It goes through some sort of driver, then to the CPU, I dont know what is required to make it work.

@gregorg , I have not been able to get the Buzzer working here either.
Also, were you testing your voltage scale through your PDB or from your battery? there is voltage drop on a PDB if you have a video filter in place.
I have not used telemetry, since I am using OSD, Blackbox and RX serial.....More serial ports needed.... I might look into softserial.....

gregorg, 2016-07-02:
I have a pdb with a filter, however I was testing directly with a capacity controller on the JST-XH lipo pins.

Getting the buzzer to work is an important feature, I understand the lack of information, I will ask Felix directly.

llambkin, 2016-07-02:
VFAS is working for me here, on UART 1, the AUX pin it was on on KISS, is not assigned as a UART.

martinbudden, 2016-07-02:
@gregorg , I've updated voltage scale to 160 and set serial RX to use UART2 by default.

Can you test and make sure it is all OK?

dgddustin, 2016-07-02:
Okay so I have my kissed alien 5in flying great but one thing I have noticed is when I go punch out high in the sky and hold full right or left roll to flip it will do just that but if I keep holding it and trying to keep flipping like 3 to 5+ times it's starts slowing down dramasticly almost to stopping. It might stop but I haven't let it get that far before I release roll and everything is fine again. It's like pulling a starter string on a lawnmower and the further you pull the more resistance you get. Does anyone know what might cause that?

borisbstyle, 2016-07-03:
Added in 3.0!

closed 2016-07-03 by borisbstyle

schuerni, 2017-02-02:
Hey there... I'm running BF on my KIss, superthing!, But how when does LED support Ws2812 work?
THx schuerni

[9 more replies]