betaflight/betaflight #179: transponder spf3mini

dannygravity, 2016-02-11:
transponder on spf3mini is not working with this firmware.

borisbstyle, 2016-02-11:
@dannygravity 
Yes correct. I know that. Not a priority for now.

borisbstyle, 2016-03-03:
This should be resolved in 2.5

timdonovanuk, 2016-03-24:
Hi Boris, everyone is reporting [here](http://www.rcgroups.com/forums/showthread.php?t=2592215&page=29#post34345533) that the transponders are not working in the latest BF. Could you please confirm its definitely resolved? Cheers!

borisbstyle, 2016-03-25:
@timdonovanuk It has never been tested. I would not be surprised that it doesnt work actually.
Use cleanflight if you want to use transponder as I really dont have oppertunity and tools for transponder testing.

borisbstyle, 2016-03-25:
I will reopen the issue so people are aware that it doesnt work

timdonovanuk, 2016-03-26:
Great, thanks for clarifying! Cheers

rigoneri, 2016-04-06:
Noticed that the transponder code in main.c is commented out since you merged changes on Feb 1st/2nd... Maybe that's why it's not working? https://github.com/borisbstyle/betaflight/blob/betaflight/src/main/main.c#L537

Here's the merge commit: https://github.com/borisbstyle/betaflight/commit/220850e429793c2c87d2a1ea16b3c9fc0f19176c#diff-c27ac60319310f0e3b936bca81499818

rigoneri, 2016-04-06:
Also looking at the cleanflight changes for the SPRacingF3Mini I noticed that `serial_cli.c` is missing `TRANSPONDER` from the list of `featureNames[]`, not sure what that does but probably not a big deal...

Cleanflight: https://github.com/cleanflight/cleanflight/blob/master/src/main/io/serial_cli.c#L197
Betaflight: https://github.com/borisbstyle/betaflight/blob/betaflight/src/main/io/serial_cli.c#L196

borisbstyle, 2016-04-06:
Yes it is commented as it didnt work and was conflicting and braking the
target completaly. I put a comment to fix it in the future, but totally
down on the priority list at the moment

rigoneri, 2016-04-07:
Ok seems that the main reason the target wasn't building was that `drivers/usb_io.c ` was missing from the Makefile under `VCP_SRC`. I haven't tested if it works yet.

PR #329 (2016-04-07): Fixed SPRacingF3Mini transponder
Fixed issue that was causing the SPRacingF3Mini target not to build when transponder code was uncommented out. Fixes #179

Btw I've tested that the transponder works after these changes. Thanks!
commit d6550827 (2016-04-07): Fixed issue that was causing the SPRacingF3Mini target not to build when transponder code was uncommented out. Fixes #179
closed 2016-04-07 by borisbstyle