betaflight/betaflight #183: Voltage sensor reading

Cheredanine, 2016-02-11:
Using a lux, not tried on other FCs, 2.4 RC7 with props off and no tx input. Voltage in cf setup page varies by 1v+ (current sat with a 4s battery plugger in reads from 15v to 16.1, rapidly oscillating back and forth. (When flying getting low voltage warnings way too soon appearing very briefly with this pack at this voltage)
Flashed with CF (colibri target) 1.2 rc2 and same issue appears.

Flashed with BF 3.2.5 and voltage stays steady at 15.6v
Flashed with CF 1.11 (colibri target) and voltage stays steady at 15.6v

(Not raised issue in CF section, clearly an issue for PID scaling to voltage)

mikeller, 2016-02-11:
I can confirm that this is happening with a Naze32 rev 6 and betaflight 2.4.0-RC7 as well.

I got random and sporadic battery warnings when flying. Flashed back to 2.4.0-RC6, with the same battery (never removed from quad), and the problems are gone.

Cheredanine, 2016-02-11:
Confirmed on lux, rc6 doesn't have the issue

gerro1977rm, 2016-02-12:
Same thing happens on a CC3D with beatflight 2.4.0 rc7 as well.

borisbstyle, 2016-02-12:
anybody has a log?

mikeller, 2016-02-12:
Hi @borisbstyle.

On 12 Feb 2016 7:02 pm, "borisbstyle" <notifications@github.com> wrote:
>
> anybody has a log?

I can generate one tonight.

borisbstyle, 2016-02-12:
@Cheredanine 
This is not an issue in PID scaling to voltage feature. This is something else. RC6 also had the voltage scaling and that works

mikeller, 2016-02-12:
Hi @borisbstyle.

Got a log for you. It has the first voltage warning about 30s in, then keeps on going on and off for a couple of minutes.

https://drive.google.com/file/d/0B_6c1X6V1qkAeENHQi1HQVQ0OWM/view?usp=sharing

Hope that helps shed some light on what's going on.

Cheredanine, 2016-02-12:
@borisbstyle sorry should have been clearer, I didn't mean it was an issue IN PID scaling, I meant it is an issue FOR PID scaling

readerror67, 2016-02-12:
Testing with a Lux and RC7 on the bench w/ a lab PSU, shows steady at 15.2V w/o any fluctuation, maybe something different happens when armed though.

Cheredanine, 2016-02-12:
@readerror67 didn't need to arm, when I was testing just literally flashed, connected and plug in lipo.

mikeller, 2016-02-12:
Hi @readerror67.

Probably. I only ever seen it happen when armed, or after disarming after flying.

readerror67, 2016-02-12:
There were a few changes to the battery ADC stuff in the past few days...
https://github.com/borisbstyle/betaflight/commit/27772ba56aafd0aa0497280472e4e2ade9bc8262#diff-ba13bb2051b35b7f32675cbeefd2f2fbR79

https://github.com/borisbstyle/betaflight/commit/83d8a8441c7db6aea6ff669e7cf1f6464a1e5022#diff-ba13bb2051b35b7f32675cbeefd2f2fbR72

Odd that the issues are all across the board though, some armed/flying and others on the bench and some, no issues.

I can't hook up a debugger right now but also can't replicate the issue so unsure how much help I would be.

Cheredanine, 2016-02-12:
I noticed it whilst flying due to the low battery alarm mate, suspect it is always there, just didn't notice in config when first flashed
Assume your lab supply isn't a lipo pack, suggest you try with one

readerror67, 2016-02-12:
DE just commented:
https://github.com/borisbstyle/betaflight/commit/bb09f5da2b10e381af98565241b5dfe16aba6e4d#commitcomment-16054944

borisbstyle, 2016-02-12:
Fixed here
https://github.com/borisbstyle/betaflight/commit/49ab107e2607ea84cc833edacecdc91b22244cb1

Cheredanine, 2016-02-12:
Want this closed?

borisbstyle, 2016-02-12:
I will first release RC8 and close it

Cheredanine, 2016-02-12:
Cool, will leave to you, assume it is being added to cf 1.12 rc2

borisbstyle, 2016-02-12:
I made a PR to cf1.12

Cheredanine, 2016-02-12:
@borisbstyle RC8 7/10 mate, it has dramatically reduced (current fluctuation is between 16.3-16.7V on default scaling, but clearly still there, flashed back to RC6 and voltage reading is stable and solid, flashed back to RC8 and still all over the shop, just over a smaller range than RC7

borisbstyle, 2016-02-12:
@Cheredanine 
Can you post a log of it. Battery can fluctuate a bit of course. It's about how it holds during flight

Cheredanine, 2016-02-12:
@borisbstyle you mean blackbox? I dont have a bblogger :( if I can provide something else - tethered logging, video etc, happy to)

borisbstyle, 2016-02-12:
vbat isnt really stable......as long as it doesnt go down to 1volt etc that should be fine. 16,2-16,3-16,8 its all fine.

In the RC7 it would go down to 1V etc

Cheredanine, 2016-02-12:
no, in RC7 the upper and lower limits of its fluctuation was 1V appart (with the pack I used for bench testing it would vary from 15.0 to 16.1V or at least that was what was showing when using an MSP connection (MWOSD or CF Configurator)

in RC8 it is jumping up and down (on a charged pack) between 16.3 and 16.7 reasonably quickly (it doesnt stay still)

in RC6 and all earlier releases it sits still on a reading

borisbstyle, 2016-02-12:
@Cheredanine 
Ok glad to hear that anyway that the jumps are just that small. That can be easily fixed

Cheredanine, 2016-02-12:
https://onedrive.live.com/redir?resid=5957BA8F0800A782!1630&authkey=!AOyrvhy9gWVG_RE&ithint=file%2cMOV

hope that works, it should be a vid so you can see behavior

borisbstyle, 2016-02-12:
@Cheredanine 
Can you download this file and rename to .hex and see how it goes
[image]

Cheredanine, 2016-02-12:
same, (16.1 to 6.5V)

borisbstyle, 2016-02-12:
Can you try this one. Not different, but it has debugging added. Go to sensors tab and click on debug. THere you can see debug 2 and debug 3. Tell me what it does and post a screen. Also speed up the reading to 10ms

[image]

Cheredanine, 2016-02-12:
behavior as described above and shown in video,
debug screen capture attached
[image]

borisbstyle, 2016-02-12:
Looks actually stable to me

Cheredanine, 2016-02-12:
and yet it is moving up and down by about half a volt (see vid)
https://onedrive.live.com/redir?resid=5957BA8F0800A782!1632&authkey=!AG8u14PeE7Ip6ic&ithint=video%2cmov

borisbstyle, 2016-02-12:
The line still looks pretty flat. I would just fly and see how it goes

Cheredanine, 2016-02-12:
happy flying it mate, was happy flying rc7. I know how long my lipos last cut a little time off in case power draw increased.

it is going to cause flakey voltage alarms and it is going to get comments from the liveware because there was a stable voltage reading in cf 1.11 and BF2.4 rc6 and earlier.

bear in mind the fluctuation is not under load with in flight voltage sag it will be a bigger variation.
I can not longer glance at my voltage reading in my osd and know what my voltage is, I have to watch it for a few seconds and figure out the average,

it is of course your choice to close or not and if not how much effort to put in to resolution :) will be flying rc8 at lunchtime shortly

borisbstyle, 2016-02-12:
I am not saying that i will leave it but its good for now. I will test when
I get home.

rc7 had a bigger underlaying problem what needed an immediate fix.

borisbstyle, 2016-02-12:
can you try this one?
[image]

Cheredanine, 2016-02-12:
well it was different, voltage constant at 1.3V (on a 4S)

borisbstyle, 2016-02-12:
oh....lol. I think i sent you a wrong hex. Will send a new one.

borisbstyle, 2016-02-12:
[image]

Cheredanine, 2016-02-12:
better, getting mostly 16.4/16.5 with variance from 16.2 to 16.7

Cheredanine, 2016-02-12:
For info Domonic asked me to raise PR in CF: 1817

borisbstyle, 2016-02-13:
is this an issue in cf 1.12 too?

digifaps, 2016-02-13:
hi , I fly the LUX , not really doable , i get a flight time of about 2 minutes , wat 5 before, mainly due to VBAT going low in the first minute, can really punch of i have voltage message on my OSD.

I'm running RC8

Cheredanine, 2016-02-13:
@borisbstyle in 1.12 rc2 yes, not in 1.11 and did not check earlier RC of 1.12

digifaps, 2016-02-14:
Hi Boris, can it be that the link you posted up higher isn't working anymore ?

borisbstyle, 2016-02-15:
On the last RC9 my Vbat voltage is rock stable. When not flying it doesnt even fluctuate.

I tested this on naze32, Dodo and SpracingF3

digifaps, 2016-02-15:
yes can confirm , much better now. thnx !

borisbstyle, 2016-02-15:
I will close this issue. I know @Cheredanine has some higher fluctuations, but I for sure don't have it anymore.

closed 2016-02-15 by borisbstyle

oliv98, 2016-02-22:
I just flashed with latest 2.4.1 and still seeing the issue.about 1 volt error max, fluctuating, not event arming. Using Naze32 rev6

digifaps, 2016-02-22:
Hi, I have to admit it is nowhere near the old VBat monitorring , somehow it seems very twitchy , and it indeed still fluctuates a bit. 

If i can help out with some tests I'll be happy to do so.

oliv98, 2016-02-22:
yes this is annoying and thus reducing flight time

borisbstyle, 2016-02-22:
its rock stable on my copters!

I dont know but maybe if you are using f1 boards and low looptimes with
many features enabled that the battery task doesnt get enough attention

oliv98, 2016-02-22:
using a naze32 rev6, pretty default settings, no gps,default looptime,no leds, lux mode

oliv98, 2016-02-22:
and my external voltmeter showing steady voltage.
could be the physical sensor?

digifaps, 2016-02-22:
I'm using a LUX boris , so it's F3 running on 2khz

borisbstyle, 2016-02-22:
I verified this on Dodo, spfracingf3 and Naze32 rev5. It really stays
stable with 0.1 volt fluctuation at max

borisbstyle, 2016-02-22:
cheradanine also verified on LUX_RACE board

digifaps, 2016-02-22:
It's not only when idle , but for example punch-outs are almost undoable as I am paranoid about Voltage levels, after switching lipo I have about 15.1V left, yet when flying i get alarms every 5 seconds.

how can I help to pin point this ?

borisbstyle, 2016-02-22:
during punch out it is perfectly normal that the lipo will sag :)

If you have an osd you can watch that live

digifaps, 2016-02-22:
boris, I have more than one quad, and i have flow this same model with other versions of betaflight and cleanflight. I do fly with an OSD and i really can see a difference in behaviour.

please don't take me as a beginner , i have quite some arduino, embedded and electronics experience.

[54 more replies]