betaflight/betaflight #191: Airmode switch on spf3 / 2.4.0 no function? and reinforce the airmode?

cybexz, 2016-02-13:
the airmode on the switch has no function on switching . Can I reinforce the airmode ? 
with set airmode_saturation......?

borisbstyle, 2016-02-13:
how do you mean it has no function?

It works like a idle up switch in 2.4

cybexz, 2016-02-13:
when i switch it, i have no gain as in 2.3.5.the gain at zero gas remains the same.

borisbstyle, 2016-02-13:
in 2.4 airmode mixer is always active. you wont notice that much
difference. Only without iterm will be reset on 0 throttle and motor stop
will be overruled

cybexz, 2016-02-13:
ok i will test it asap. can i reinforce the airmode?
stronger or weaker?

borisbstyle, 2016-02-13:
airmode will always work the same when enabled

with saturation limit it gets disabled during crashes and very hard
manouvres

cybexz, 2016-02-13:
when i armed with a switch and motor stop is disabled, i dont need the airmode switch correct?
i have test it with two switches and motor stop enable, but after the motor spinns few seconds my fc beeps.

borisbstyle, 2016-02-17:
just set airmode to always enabled. It is also possible....

closed 2016-02-17 by borisbstyle