betaflight/betaflight #192: VBat still not working

cylstudio, 2016-02-13:
Frsky - X4R
Taranis X9D
It works with 2.3.5
not 2.4 RC8

waslater, 2016-02-13:
Both of mine work!

SoulPictures, 2016-02-13:
Had the same error today on the field after trying to enable 2kHz mode, voltage always has shown 0.0V 

The command I used was set gyro_lpf = off (as I was used to from earlier version) result was no Telemetry and 250-350 loop time with no arming possible (I know I used it wrong, just the telemetry problem occurred this way also on my system)

borisbstyle, 2016-02-13:
thats normal as when going for faster looptimes the system will start
dropping tasks with less priority.

besides gyro_lpf = off there is also gyro_sync_denom you have to set.

for 2khz its a value of 4.

cylstudio, 2016-02-13:
problem solved, thanks!

closed 2016-02-15 by borisbstyle