betaflight/betaflight #194: SP3 v2.4.0 modes not saving

cowboychris, 2016-02-15:
After upgrading to 2.4.0 the modes will not save after a disconnect from the USB.  I saved before disconnecting and all appeared fine, but the board would not arm.  I reconnected and found all the modes that were previously saved were now blank.  I tried to reboot to the previous release, but the boot loader was not found.  Jumping the boot loader pins did work and I was able to load the previous release, but am having the same issue - mode values will not save after the USB is disconnected.

freddiebd, 2016-02-16:
had this too- make sure your modes do not go above 2000 and they will save normally :)

borisbstyle, 2016-02-17:
I tested this and it works

closed 2016-02-17 by borisbstyle