betaflight/betaflight #196: Failsafe problem

npmullins, 2016-02-15:
Setting arm to aux3 channel and then in failsafe menu setting the aux 3 channel to move to value out of range of motor arming. Works correctly in transmitter sliders section of cf gui. But when try with battery power to quadcopter and then turn off the taranis radio the motors do not stop spinning. (using latest Lux hex target on Lux fc)

Yamaford, 2016-02-16:
You do have your rx set to no pulse right?  I do't have a lux but I just checked on the NAZE and it still works.

npmullins, 2016-02-16:
I have my 4xr RX failsafe set to zero throttle and aux 3 set to motors not armed. I will try a test with battery power and then I will disconnect the RX and see what happens.

borisbstyle, 2016-02-16:
you need to set no pulses for failsafe and not use failsafe from the rx

npmullins, 2016-02-16:
i have now activated stage 2 failsafe in the new cf failsafe menu and it now is working.

closed 2016-02-16 by npmullins