betaflight/betaflight #206: Incorporate Raceflight into Betaflight

opiswahn, 2016-02-22:
***Disclaimer***
Don't get me wrong, I don't want to start some kind of war here, this is just what i belive would be best for the community.

It would be nice to habe the raceflight changes available in Betaflight. As far as i know, raceflight's only purpose is to have F4 support.
The main problem with that is, that raceflight lags the feature availability from betaflight or even cleanflight. The codebase is in part way older then betaflight or cleanflight, and the porting is very slow if ever possible. This is also due to the issue that raceflight isn't a true fork of betaflight (in GIT terms), so up- or downporting of features is unnessessary complicated.

I really appetiate rs2k for it's work to supply F4's very much, but i think it would serve the community better to have one version supporting all targets with a united dev-power.
That means better testing, better code quality, more devs working for a united goal.

petitssoldats, 2016-02-22:
:+1: +1 

And bring multishot support to betaflight 🚀

borisbstyle, 2016-02-23:
@petitssoldats 
Multishot will be in betaflight.

I will close this issue as the discussion was carried in raceflight github

closed 2016-02-23 by borisbstyle

petitssoldats, 2016-02-28:
@borisbstyle  
:heart: _ :heart: