betaflight/betaflight #209: Motor output error.

Mikael-Jakhelln, 2016-02-23:
@  roll/pitch input <950 && >1950, motors spaz out and slow down.
& Idle motor output is higher than minimum throttle, and causes liftoff @ 0% throttle.

On naze/flip32, using ppm with arm switch, both with and without airmode, no acro+.
Rc inputs show correct values in cf-config, no spazzing here. No problems running motors from gui.
Tested on multiple boards.

Issue present on 2.4.0 and 2.4.1, but not on 2.3.0.

borisbstyle, 2016-02-23:
When you input throttle higher than min_check the iterm will wind up, but even without it it will try to correct on the ground. 
Especially when there is some oscillation it will even spool up more

Mikael-Jakhelln, 2016-02-23:
Liftoff happens when input throttle is lower than min_check, arming via switch, airmode is off.

throttle input = 1000
min_check=1100
min_throttle=1040

Same settings works in 2.3 and below. Liftoff is achieved at around 1200 when working properly.

High roll/pitch value issues happen regardless of throttle value, and always when sub/exceeding the same values. Is anything new triggering @ 95% throttle in 2.4+?

borisbstyle, 2016-02-23:
lower min_check and configure your min_throttle as low as possible.

The older version was not using rcCommand and thats why the low throttle
was lower and you were also missing your turottle curve etc.

rcCommand has a lower limit which is the min check.

borisbstyle, 2016-02-23:
its all in the wiki

Mikael-Jakhelln, 2016-02-24:
Thanks.

So I tried that. And read the wiki a couple more times.

Flashed back to 2.4.1, Re-calibrated esc's and set min_throttle is as low as it can go before getting motor jitter, 1038. Lowered min_check so low it barely arms. And set to arm by switch. Everything else is default. Idle rpm seem ok now.

But the main problem is the extreme sinusoidal fluctuations in motor outputs. Pointing to default pid's/settings going haywire.
I tried dulling down P's and I's, but outputs still fluctuate wildly. Looked into D-smoothing & filtering settings.

Setting gyro_lpf from off to 188hz reduced fluctuations. It's still a bit brutal, but it sounds flyable. Setting it lower does not smooth things further.

Thoughs?

laurynas-karvelis, 2016-02-25:
Hi @Mikael-Jakhelln, I had a similar issue with my quad Naze32 rev6, Betaflight 2.4.0, and **4S battery**, happens ONLY with 4S, with 3S it lifts off perfectly. What I had to do to mask this issue was to lower my P settings for both roll/pitch and yaw. Also, had to lower mid-throttle to 0.10 (default 0.5)! Only then the issues started to disappear.

Betaflight 2.4.1 handles lift off better, but still I have to be very careful when increasing throttle while still on the ground.

borisbstyle, 2016-02-25:
decrease min_check

Mikael-Jakhelln, 2016-02-25:
I'll try lowering mid throttle. But it's not the zero throttle liftoff that's the biggest problem. It seems to be a symptom of something else.

The major motor output fluctuations is the main problem. Since the only thing that reduced the oscillations was setting gyro_lpf_hz=188hz, it seems the issue is **connected to gyro/D filtering?**

I already tried lowering roll/pitch/yaw pids, issue is present even at ridiculously low settings.

laurynas-karvelis, 2016-02-25:
Thank you, @borisbstyle, it helped, though more people could experience this issue. Just wondering if it would be possible to check on the f/w if `mix_check` is higher than `min_throttle`, if so, `min_check` becomes at least equal value to `min_throttle` once armed.

The best solution I see to avoid such user issues would be within cleanflight configurator, isn't it? Any thoughts?

borisbstyle, 2016-02-27:
try also 2.5 where min_check doesnt restrict throttle

closed 2016-02-27 by borisbstyle