betaflight/betaflight #210: Vbat fluctuation on LUX_RACE targets with 2Khz setting

digifaps, 2016-02-23:
when I run my LUX FC in 2Khz mode i experience voltage sags on my VBAT monitor of about 1V. this happens when going form idle to full throttle when props are off.

on 1 Khz I experience only half the voltage sag.

Video that shows the behaviour :

https://www.youtube.com/watch?v=lTI5CycpWag

the first part is at looptime 1000

you see me switch to looptime 500 max voltage sag = 0.5v
but only on 2.00min i have the 4 motors up and running. (you'll hear the esc signal)
then the max voltage sag is 1.0v

besides that , when switching to 2Khz the Vbat varies 0.3v and on 1Khz it's 0.1v

To me it clearly shows a difference between the two.

When using an OSD the voltage reading becomes so hectic you can't even make up where you are with it.

Highoctane32, 2016-02-24:
I noticed no unusual VBAT readings / voltage sag running 2.4.1 at 4khz on the Lux today. Checked it at idle with props ON .1-.2 sag which matched my OSD reading within .1v (RROSD with standalone voltage/current sensor). At moderate throttle, again props on, I saw a max of .3v drop which also matched the OSD current sensor. Not a complete replication since I was running 4khz. But 1) my VBAT readings were accurate within the resolution of VBAT and the OSD compared to a second source of measurement and 2) the voltage sag wasn't more than I would expect.

digifaps, 2016-02-24:
I'll try the same with 2.4.1 today

digifaps, 2016-02-24:
not powered by lipo :

Task list:
0 - SYSTEM, max = 10 us, avg = 0 us, total = 5 ms
1 - GYRO/PID, max = 187 us, avg = 135 us, total = 31134 ms
2 - ACCEL, max = 21 us, avg = 0 us, total = 163 ms
3 - SERIAL, max = 1554 us, avg = 2 us, total = 167 ms
4 - BEEPER, max = 14 us, avg = 0 us, total = 62 ms
5 - BATTERY, max = 17 us, avg = 0 us, total = 27 ms
6 - RX, max = 95 us, avg = 64 us, total = 439 ms
8 - TELEMETRY, max = 36 us, avg = 3 us, total = 450 ms

powered by lipo :

Task list:
0 - SYSTEM, max = 10 us, avg = 0 us, total = 13 ms
1 - GYRO/PID, max = 187 us, avg = 139 us, total = 73387 ms
2 - ACCEL, max = 21 us, avg = 0 us, total = 386 ms
3 - SERIAL, max = 1554 us, avg = 2 us, total = 209 ms
4 - BEEPER, max = 14 us, avg = 0 us, total = 121 ms
5 - BATTERY, max = 40065 us, avg = 2 us, total = 144 ms
6 - RX, max = 95 us, avg = 59 us, total = 1002 ms
8 - TELEMETRY, max = 36 us, avg = 3 us, total = 716 ms

digifaps, 2016-02-24:
ok now my FC is stuck in DFU mode. any ideas how to fix this ?

digifaps, 2016-02-24:
in cleanflight 1.12 is have the same behaviour as in 2.4.

borisbstyle, 2016-02-24:
Can you rename this version to .hex and tell me what the debug value 3 says from the sensors tab
[image]

Nihilistic, 2016-02-24:
Getting the same VBAT fluctuation on my Naze32 Rev5 with CleanFlight 1.12.0, in flight in 1khz mode.

Cheredanine, 2016-02-24:
@borisbstyle do you want that with/without  battery and what throttle activity?

borisbstyle, 2016-02-25:
@Cheredanine 
Just read what value you see on different settings/looptimes

digifaps, 2016-02-26:
flu knocked me down , will try to provide later today

borisbstyle, 2016-03-23:
Any update on this?

There will be a configurable hysteresis to voltage alarm in the next release anyway

Cheredanine, 2016-03-23:
Fly every day, voltage fluctuation within 0.1v sat on the ground with motors stopped, voltage sag in flight up to 0.5v but only as a result of heavy load/high throttle, all at 8khz though

borisbstyle, 2016-03-31:
try increasing the new vbat_hysteresis in 2.6

closed 2016-03-31 by borisbstyle