betaflight/betaflight #211: 2.4.1

johnnoboa, 2016-02-23:
FrSky D4R2 Naze 32 rev6   nothing works had to revert to previous release

No arming and modes didi not change

borisbstyle, 2016-02-23:
I actually have a quad with D4R2 on naze32 rev5 and it works fine. Also few others I know fly it so I dont think this issue is valid.

closed 2016-02-23 by borisbstyle

johnnoboa, 2016-02-23:
I will retry tonight thank you very much

</div><div>
</div><div>-------- Original message --------</div><div>From: borisbstyle <notifications@github.com> </div><div>Date: 2/23/2016  5:13 PM  (GMT-05:00) </div><div>To: borisbstyle/betaflight <betaflight@noreply.github.com> </div><div>Cc: johnnoboa <pro-shop24@live.com> </div><div>Subject: Re: [betaflight] 2.4.1 (#211) </div><div>
</div>
I actually have a quad with D4R2 on naze32 rev5 and it works fine. Also few others I know fly it so I dont think this issue is valid.

---
https://github.com/borisbstyle/betaflight/issues/211#issuecomment-187939780

johnnoboa, 2016-02-23:
I am good it works wonderful

</div><div>
</div><div>-------- Original message --------</div><div>From: borisbstyle <notifications@github.com> </div><div>Date: 2/23/2016  5:13 PM  (GMT-05:00) </div><div>To: borisbstyle/betaflight <betaflight@noreply.github.com> </div><div>Cc: johnnoboa <pro-shop24@live.com> </div><div>Subject: Re: [betaflight] 2.4.1 (#211) </div><div>
</div>
I actually have a quad with D4R2 on naze32 rev5 and it works fine. Also few others I know fly it so I dont think this issue is valid.

---
https://github.com/borisbstyle/betaflight/issues/211#issuecomment-187939780