betaflight/betaflight #212: Profile/rateprofile mixup

Hydroculture, 2016-02-23:
Hoi Boris!

Hello!

Playing with 2.4.1 
Using spracingf3.

1.
im having problems with the new profile/rateprofile mixup.

I used to have two profiles - one for 4s, one for 3s.
These profiles changed the PID values. Could do with a switch.
Flipping the switch changed PIDs (via adjustment tab),
and also rate profile (via default_rate_profile in profile section).
When i do it in 2.4.1, only rateprofile is changed, PIDs stay the same.

2. in cli typing 
profile 0
dump profile
- copy cli output via strg C
profile 1
- paste profile 0 via strg V in profile 1
save

after FC resets, dump is not possible anymore,
neither in CFgui, nor in EZgui. The dump output gets truncated

Problem persists, even after restarting gui and disconnecting/reconnecting FC to USB

Thank you for the hard work

best 
Holger Landwehr

KiteAnton, 2016-02-23:
For your first issue, switching profile through switch is not possible, only rates. 

For second issue: I have tested this now with 3 different boards and:
* Naze32 rev6: Works, no problem. 
* Spracingf3: Do not work, gets stuck in dump. 
* Dodo: With RMDO target: Works, no problem. With Spracingf3 target, same as for spracingf3. 

I will see if I find anything in the code that can cause this problem.

KiteAnton, 2016-02-23:
I found the problem. It is however a bit strange that it works on some targets and some not but I guess it depends on how the memory is partitioned. 
I will create a PR for this bugfix soon.

closed 2016-02-23 by borisbstyle

Hydroculture, 2016-02-24:
Thank you so much!
 Would it be possible, to share the .hex with me? Would like to test 2.4.1

Thank you
Holger

KiteAnton, 2016-02-24:
andWho has a jenkins server automatically building the latest code every 4 hours or so, so the fix is available there, http://andwho.sytes.net:8080/job/BorisB_BetaFlight/
However be aware that it is building directly from source/commits and there is a possibility that other still unknown bugs exists.

Hydroculture, 2016-02-24:
Thank you!

had a look at the wiki and found jenkins 203 build with your fix.

best
Holger

PR #213 (2016-02-23): Bug fix for profile dump hickup
Bugfix to fix #212 . The pointer magic involved with using PROFILE_VALUE caused faulty memory.