betaflight/betaflight #216: Cleanflight spins motors every time I hit save..

ghosTECH, 2016-02-25:
I have put together a 210mm quadcopter with OP CC3D/ Ipeaka 15A ONESHOT125/ 2204 2300kV motors. After I assembled parts together, I used OP to configure and test quad. It arms, spins motors, communicates with X9D+ using SBUS connection/ FrSky X4R-SB.
Went ahead and flashed Betaflight 2.4.1 since im using BF on my other CC3D quad, and tried to configure using cleanflight. I leave ONESHOT on, stop motor spin off, serial connection, sbus, min thottle at 1000, mid 1500, max 2000, rssi on. 
I am able to configure Taranis to communicate with CleanFlight, and the moment i hit save (quad is powered since it needs to be powered to get radio communication), the motors start spinning. 
If I hit disconnect and remove usb from macbook, motors continue to spin. I have to unplug battery to stop motors. When I try to calibrate ESCs, I switch test on and move master all the way up, plug in battery and ESCs only make a short sound. After that, if i try to test motors, they do not spin. The only way to make the motors spin is to hit save on Receiver tab, or Modes tab, or Configuration tab...anywhere you can hit save. 
I have had this problem for 6 days, and I have looked and tried asking on RCgroups for help. What am I missing?

Flugsicherung, 2016-02-25:
You do the motor calibration not ready. After "I switch test on and move master all the way up, plug in battery and ESCs only make a short sound." you must set master fast down and another beep from ESC´s sounds and than kalibration is ready.
That is not a bug, please close issue.

borisbstyle, 2016-02-25:
This is normal behaviour of oneshot125 pulses when flashing and doing stuff with powered on board. That is also why I dont like CC3D where board needs to be powered in order to do things.

Always remove props on those FC's!

closed 2016-02-25 by borisbstyle

ghosTECH, 2016-02-25:
First, thanks for the reply...and I apologize if this is not the right place to do this.
I have set master all the way down the moment the ESCs start to make a sound, and there is a second ESC set of sounds the ESCs make. None the less, the quad wont arm and function. In Mode tab, I do see that the arming turns green when radio and battery is connected. When i try to arm the quad, motors dont spin or have any response. Am I missing a CLI command? Perhaps I have something selected that im not supposed to? 
Im sorry, I dont know where else to look.