betaflight/betaflight #218: Beeper tone on startup can't be deactivated

lblblb, 2016-02-25:
I'm running Betaflight 2.4.1 and I can't seem to fully customise the beeper modes, it's still beeping when I plug in the LiPo, despite my tryings to diable that and other occurences of it beeping. I've used the beeper_off_flags command as explained here:
https://github.com/borisbstyle/betaflight/wiki/Betaflight-specific-CLI-commands
and applied the sum of all beepers except 512 (for the switch beeping I want to keep on). I set it to 16383−512=15871 but it's still beeping for a second when applying power, other than that it does behave as I want it to. 16383 is the sum of all relevant off flags from 1 to 8192.

borisbstyle, 2016-02-26:
isnt itthe escs beeping on startup?

lblblb, 2016-02-26:
No, both the ESCs and the buzzer are making noise on startup, I'm using the regular BZ output of the Lumenier Lux board.

borisbstyle, 2016-02-27:
These are all available commands 

BEEPER_GYRO_CALIBRATED, 1
 BEEPER_RX_LOST_LANDING, 2 // Beeps SOS when armed and TX is turned off or signal lost (autolanding/autodisarm)
 BEEPER_RX_LOST, 4         // Beeps when TX is turned off or signal lost (repeat until TX is okay)
 BEEPER_DISARMING, 8       // Beep when disarming the board
 BEEPER_ARMING, 16         // Beep when arming the board
 BEEPER_ARMING_GPS_FIX, 32 // Beep a special tone when arming the board and GPS has fix
 BEEPER_BAT_CRIT_LOW, 64   // Longer warning beeps when battery is critically low (repeats)
 BEEPER_BAT_LOW, 128       // Warning beeps when battery is getting low (repeats)
 BEEPER_, 256              // when plugged into USB
 BEEPER_RX_SET, 512        // Beeps when aux channel is set for beep or beep sequence how many satellites has found if GPS enabled
 BEEPER_DISARM_REPEAT, 1024        // Beeps sounded while stick held in disarm position
 BEEPER_ACC_CALIBRATION, 2048      // ACC inflight calibration completed confirmation
 BEEPER_ACC_CALIBRATION_FAIL, 4096 // ACC inflight calibration failed
 BEEPER_READY_BEEP, 8192           // Ring a tone when GPS is locked and ready
 BEEPER_MULTI_BEEPS, 16384         // Internal value used by 'beeperConfirmationBeeps()'.
 BEEPER_ARMED, 32768               // Warning beeps when board is armed (repeats until board is disarmed or throttle is increased)

lblblb, 2016-02-28:
^Yes, these are the commands / values I'm working with here, albeit the highest one is new to me. 

I testet it now again with the new set of beeper modes: The sum of 2^0 (1) up to 2^15 (32768) is 2^16−1, subtracting 512 for the switch beeping gives 65023. I set that, saved, powercycled and I still get a fast beeping sequence lasting maybe 0.5 seconds right after board power on. Even after setting it to 2^16−1=65535, so disabling all the beeper modes, I can't get rid of the startup beeping sequence.

borisbstyle, 2016-03-01:
I will check to add that one

Pierre-A, 2016-03-01:
**@lblb** you're right, there is an itnitialization step (in _main.c_) which has been forgotten in the list.
**@bosibstyle** Look at here https://github.com/cleanflight/cleanflight/pull/1894 , I redesigned this function, replacing the _beeper_off_flags_ parameter by a more friendly _**beeper**_ CLI command.

I'll add this last case as BEEPER_SYSTEM_INIT.

Pierre-A, 2016-03-02:
done, initialisation beeps are disabled when the BEEPER_SYSTEM_INIT bit is set in the _prefered_ configuration.

see  https://github.com/cleanflight/cleanflight/pull/1894

borisbstyle, 2016-03-03:
@Pierre-A 
Thanks....will have a look into that

Flugsicherung, 2016-03-03:
Hi together,
I use 2.5.0 RC3 on Naze32 and I have the same Problem when I connect the battery beeping goes on.
I find out, when I disconnect MV-OSD on startup and connect the OSD after battery connection everything works right. So I thing the Problem is on startup sequence for the init of serial Interfaces.

borisbstyle, 2016-03-03:
fixed in 2.5 thanks to @Pierre-A 

@Pierre-A 
I also added ON_USB beeper option. Some board hae Beeper powered when USB cable connected

closed 2016-03-03 by borisbstyle

anta377, 2016-03-07:
Hi Everybody
I put betaflight 2.5.3, when i plugged a battery beeper tone always, i try each command to see when the beeper stopped. Only one stop the beeper, i think it's beeper_armed, but no beeper for other command too, so what is the issue?
Thanks

anta377, 2016-03-07:
Sorry my flight board is a naze 32 rev 6 a

lblblb, 2016-03-07:
^ Make sure nothing else is triggering the beeper, I think you might have a voltage alarm activated that triggers it. Your issue does not fit to this reportet bug, I think it's a misconfiguration.

anta377, 2016-03-07:
nothing else is triggering the beeper, just put in cleanflight the mini, max cells. and beeper is in a 2 position inter but disable when i plugged the battery, when i plugged, removed, plugged and remove battery, sometimes all is ok and beeper don't tone

lblblb, 2016-03-07:
If you can't find the source of the behaviour, then create a new issue about it and describe everything. This issue is closed and describes a very specific, but different problem.