betaflight/betaflight #219: Cross motors accelerating

kennuard, 2016-02-25:
Hello,

I have Openpilot MINI - the "chinese" version from banggood. I have PROPER setup QUAD! There is a serious problem with last few versions of betaflight where cross-motors are accelerating. Ive seen same posts on reddit about this happening to guys....where this issue got blown away by how bad at setting it up they are. How do I know its a problem of new firmware ?

I burnt my old control board. Ive got now one from friend. Did not even start it. Quad still set up from previous desk. Flashed to newest firmware. Boom....motor problem. Flashed back to previous firmware Ive still got on desktop ( Betaflight 2.1.5 ). Quad motors seems just fine - quad flies like a dream again.

Problem description. Flashed with 2.4.1. Armed:

Lowest RX throttle - 1-4 runs on 1150, 2-3 runs on 1600-1800
Highest RX throttle - 1-4 runs on 1400, 2-3 runs on 1600-1800

Also noticed that motor 3 is beeping about 0.5 - 1 second after other motors on boot.

Tries master controll and all motors spins exactly as set by master. Seems like a roblem of Rx input combined with "something".

borisbstyle, 2016-02-25:
I think it is because the copter wants to stabilise when you arm it. I
would say that it will fly fine perhaps?

kennuard, 2016-02-26:
Its more like some sort of internal loop. When the range is 1100 - 1800. The motor jumps right to 1200 and then slowly accelerates towards 1800. 

Its also weird that in hand (I know its not a 100% valid test) when I tried to go in direction with the 3 motor - it wouldnt loose the RPM. The board seems to ignore the accelerometer/gyro input on that direction. 

I guess its just a CC3D problem. And I am more of a safety guy and if it wont hover - I wont even try to fly it normaly. And most of the time - on min. throttle - the quad is jumping like a frog.

borisbstyle, 2016-02-26:
testing on the ground without props can produce unexpected behaviour as
everything what pid controller wants to do doesnt work out.

kennuard, 2016-02-26:
The quad is uncontrollable with props and the lowest throttle input possible. As far as I can tell It wont hover. From what Ive experienced it starts jumping in a direction of 1 o'clock and no input can stabilise it. In this case its dangerous to try to fly it :)

As I said. I am using standard (6dof) 15$ Openpilot mini board from banggood. If you need another data - I can try to reproduce it and get some logs. Other than that Ill stick to the older version as its sufficient for my needs.

borisbstyle, 2016-02-27:
I will close this issue as it is confusing, but nobody else has the issue.

closed 2016-02-27 by borisbstyle

kennuard, 2016-02-29:
https://www.reddit.com/r/Multicopter/comments/3u5ajm/motor_accelerationdeceleration_issue/