betaflight/betaflight #221: Beep Buzzer after arm

ixiliar, 2016-02-25:
Hello! I have a problem to see if someone has pasaado him. Anyway, I've updated the Lisam 210 and Lisam 180, both with naze32 rev6, and now when I build, is in the way this the buzzer beeps every two or three seconds, is no battery because the vbat I have reviewed and the battery was full and I've tried several, also happens to me in both. The buzzer stops ringing at the time of the throttle. Thank you!!

KiteAnton, 2016-02-26:
You don't mention anything about your RX but I would guess it beeps because it is not recognized. 
Try set beeper_off_flags = 4. That will disable beeper for lost RX. If that not helps then I suggest you read up on the beeper of flags on the wiki and disabling them one by one to debug what's causing the beeps.

ixiliar, 2016-02-26:
My EX is a FRSKY D4RII with a TARANIS. Connected by PPM.

borisbstyle, 2016-02-27:
closing as its described in wiki

closed 2016-02-27 by borisbstyle