betaflight/betaflight #223: Vortex 285 OSD - PID tuning not possible

timdonovanuk, 2016-02-26:
Hi,

I thought the PID API was generic, which is why apps that have no specific knowledge of BF but only CF can still access the board to make PID changes (e.g. EZ-GUI, minimosd etc.). However on the latest BF (2.4.1), the Vortex 285 OSD gives the error "Incompatible FC API" when trying to make PID changes. It seems to work in earlier version - has something changed here?

Many thanks!

ixiliar, 2016-02-26:
I have the same problem.

borisbstyle, 2016-02-26:
API is different through versions and the OSD / configurator always have to
check the api version to know how to talk eith it.
API van never be generic as the set parameters can differ.

timdonovanuk, 2016-02-26:
Ah I see! Fair enough. Is there anything I can do to help? I will go through the previous BF versions and see which release stops working with it.

borisbstyle, 2016-02-26:
@timdonovanuk 
There is no need for that. This is not a bug. This is just a OSD firmware on Vortex what needs upgrade from immersion side to catch up with the latest API version.

borisbstyle, 2016-02-26:
Here is the Beta version of the new OSD. Use on own risk of course :d
https://dl.dropboxusercontent.com/u/31537757/OSD_v1.0.0.25.fw

closed 2016-02-26 by borisbstyle

timdonovanuk, 2016-02-26:
Understood, wasn't sure before. Cheers Boris! I think that FW might only work on the newer Vortex Pro but I'll give it a go for sure. Thanks!

timdonovanuk, 2016-02-26:
Just in case other people find themselves here, it looks like the older [Betaflight version 2.3.5](https://github.com/borisbstyle/betaflight/releases/tag/v2.3.5) works with latest Vortex 285 Firmware (v1.0.0.21) okay.

I'm not brave enough to try flashing the v1.0.0.25 Vortex firmware, as I'm pretty sure its only for the Vortex **Pro** and not the 285.

We have to wait for ImmersionRC to update the Vortex 285 OSD firmware to be compatible. My suggestion: go remind them on Facebook that 285 owners still exist...

timdonovanuk, 2016-02-26:
Update 3...last one I promise, sorry. @borisbstyle you're a god amongst men. I bit the bullet and installed the beta Vortex OSD  fw in your link (1.0.0.25) and it works! I can't even find it on the IRC site, so who knows where its from, but it works. Thanks man :+1:

fotoingo, 2016-06-01:
Sorry to open  this thread again, but can someone point out which BF Version is usable with the latest Osd 1.28? 
I would love to try betaflight on my Vortex 285. Im using it on my Pro 250 and I like it a lot... 😁