betaflight/betaflight #225: Gyro with some very erratic behavior

timmmmmmmmm, 2016-02-26:
I fly with a Naze32 Rev6 with the latest Betaflight, and just like I had with 2.3.0 my gyro seems to get totally out of sync after a while. I'm running an OSD which displays a horizon-bar, and this shows the same strange behavior with sometimes a 45 tilted horizon while I'm flying horizontally.
Is there some feature in Betaflight that might cause some problems with the gyro or should I look into other options?

borisbstyle, 2016-02-27:
You mean the acc?

Try latest releases first

timmmmmmmmm, 2016-02-27:
Yeah could be the acc, thought that gyro was responsible for determining the horizon. I'm running the latest release.

R91g, 2016-02-27:
Hi,
I also have the same issue with original Naze32 rev6 in the Horizon mode.
Acro mode seems to work perfectly.
Tested on Betaflight 2.3.5 and Betaflight 2.4.1

borisbstyle, 2016-03-03:
Perhaps related to this problem
https://github.com/borisbstyle/betaflight/issues/239

borisbstyle, 2016-03-07:
Can someone try this on 2.5.3?

MasterZap, 2016-03-14:
I had this on 2.5.3 too, on my recent test on MW23 controller. After some happy flipping and rolling (in Acro) my artificial horizon is completely askew (like at a 45 degree angle though I am level), and it seems to hold like this even if I try to wait for a while (e.g. it doesn't seem to rectify over time).

So something is amiss. No blackbox or nuttin' sorry.

/Z

borisbstyle, 2016-03-14:
Need to ask to @digitalentity
Do you think that ignoring of some high acc readings should be ignored in order to prevent these issues?

MasterZap, 2016-03-14:
Maybe. Not sure.

I haven't quite figure out the "why" of why it happens, I've just seen it happen. (And once I didn't see it, and did my standard "switch to horizon mode to land* and it ended up on a massive angle I had to almost do full-stick to counteract, so it can be quite dangerous.

It has happened on 2 different machines, and one of them wasn't even running betaflight but CleanFlight 1.12 !! If that gives you any hints?

I'm gonna do some more tests and will report back if I figure something out.

/Z

* = yeah yeah call me a noob, but when you can hardly see something with the camera angle it's a useful feature

aricblunk, 2016-03-23:
I can tell you that my accelerometer will slowly drift out of alignment in flight, due to vibration. If I mount my FC on some double-sided tape, then the vibrations are filtered and the accelerometer doesn't drift. Just be careful to get your FC perfectly aligned before mounting.

borisbstyle, 2016-03-30:
This might minimize vibrations on 1khz looptimes for level fliers
set gyro_lpf = 188hz

closed 2016-03-30 by borisbstyle