betaflight/betaflight #227: NAZE Rev.6a ESC Cal. issue

gsx8299, 2016-02-28:
Can not calibrate ESC and min throttle is somewhere around 1350 now rolled back to last version no issues. Rev.6b does not seem to have this issue

borisbstyle, 2016-02-28:
Try RC3 version

gsx8299, 2016-02-28:
I did :( still no go

borisbstyle, 2016-02-28:
i think thats a known problem on rev6a.....thats why there is a rev6b.

It actually powers up the escs even when connected to USB....you could
calibrate it when disconnecting the esc first

gsx8299, 2016-02-28:
Yea it is ;) but...... I had thought you had fixed it because after 2.1.5 ESC cal. And PASSTHROUGH had started working normal again on the NAZE and DODO I had the same ESC Cal. Issue until 2.1.5 wish I could help more if there is anything I can do to help please let me know.

borisbstyle, 2016-02-28:
it is actually really a fc problem :)
Nothing can be done on that.....that it worked on 2.1.5 is actually
accidental.

You noe need to disconnect and connect esc servo when calibrating those fcs
that power up escs on usb

gsx8299, 2016-02-28:
Ok I'll just stay on 2.4.1 for now was hoping on trying 250. I dont know what the issue is but when I update for some reason throttle starts around 1350-1400 and I only get about 1/2 throttle after that it's weird even when I got to motor tab the motors start way later on 2.5 RC 1-3 like I would need my Endpoints to go from 1400 to 2400 now for throttle

gsx8299, 2016-02-28:
ESC cal. Aside rev.6a does not work correctly on 2.5 I swapped with rev.6b and all works normal on 2.5 rc 1-3 thx for the help

closed 2016-03-01 by borisbstyle