betaflight/betaflight #230: Lux no heading

gruntwerkz, 2016-02-29:
When I flash to the 2.5 then connect to cleanflight it doesn't show a compass heading anymore or show me moving the FC around... that tab is just frozen.

Is this normal or is it a known issue? I get that just because it's not moving in clean flight doesn't mean the board its self doesn't know.

Highoctane32, 2016-02-29:
If you set 125 looptime then the accelerometer will be turned off by default and thus you will not get that data in Cleanflight configurator. You can manually enable the acc with the CLI command set acc_hardware = AUTO if you need the acc. Make sure you check CPU usage via status command. Should be ok I think with RC3

gruntwerkz, 2016-02-29:
Loop time is 1000, the sensors all work in sensors tab shows it moving and all.

It just says my heading is 0 roll 0 pitch 0 and doesn't show the board moving at all. It seems like the board still knows it's moving according to sensor tabs just wanted to make sure it was safe to proceed.

Highoctane32, 2016-02-29:
Is the acc icon lit up?
It should be fine. But as always bench test with no props!!! It's easy to figure out if the FC is working properly in this aspect. No props, power it up and arm it and pitch, roll and yaw it by hand and make sure the appropriate motors speed up :)

Highoctane32, 2016-02-29:
Obviously it sounds like the acc is disabled, that's what gives you those readings in CFC so you won't have any self leveling if that's the case, but the gyro will still be working so rate mode should be fine. Not sure why the acc would be off at 1000 looptime, but again, you should be able to turn it on in the manner I stated above

d00bld0ze, 2016-02-29:
Same problem for me on SP racing F3. Everything seem ok, but the 3d picture is frozen and Heading/Pitch/Roll show 0° on the setup tab

Highoctane32, 2016-02-29:
Is the accelerometer icon green or blacked out?

d00bld0ze, 2016-02-29:
Acc is green and evrithing is ok in Sensor tab

Highoctane32, 2016-02-29:
Hmm..that is odd I just enabled my acc on my Lux running 2.5.0 rc3 at 8khz and all is fine in CFC, heading pitch and roll respond appropriately. Checked my Naze rev6 also running 2.5.0 rc3, lowered loop to 1000 and the acc was enabled and all the indications in CFC work fine

Highoctane32, 2016-02-29:
Should have said I raised loop on the Naze to 1000, was running 250(4khz). Anyway on both the Naze and the lux when the acc is green I get appropriate response from the pitch /roll / heading and the graphic on the set up screen..with 2.5.0 rc3

d00bld0ze, 2016-02-29:
Re flashed RC4, stock config (looptime 1000), the 3D model still blocked

Highoctane32, 2016-02-29:
Ok, I'm. Grasping at straws but I don't think it's a betaflight issue..I mean not rc3 anyway, I checked 2 different targets one of which is a Lux which was the original target for this issue and I see no such problems. What version of CFC are y'all running? On Windows (version)or Mac or what?

Highoctane32, 2016-02-29:
Ah..Y'all are both using RC4? If so is there a way to know that? If so how? :)

stephenwilley, 2016-02-29:
Pretty sure the issue is here: https://github.com/borisbstyle/betaflight/commit/6e96adff0d22d46427bb10794b1615a88a512065#commitcomment-16380537

subTasksMainPidLoop isn't getting run which includes the mag, acc, baro, blackbox etc updates

I'd submit a PR but it's just a two character comment that needs removing (I think) so a commit amend is probably more appropriate.

jimparker, 2016-02-29:
This issue is also causing horizon mode to not work. Running cc3d board

stephenwilley, 2016-02-29:
Horizon needs accelerometer, which is in subTasksMainPidLoop

jimparker, 2016-02-29:
Can I run multishot on this FC. Flycolor 30a are multishot capable with firmware

gruntwerkz, 2016-02-29:
Sorry for late reply was busy.

It is RC 4 yes, it looks like the acc icon is green, also looks like the acc is responding in sensor tab.

I did not touch the loop time, I did a  chip erase when installing the new FW as well.

The board isn't hooked up to anything so I cannot confirm if it's going to angle mode or not, I didn't wanna hook up and have issues so decided to check here first.

So yeah the whole tab is frozen, no heading all 0s no voltage or anything.

I'm using CF 1.21 I believe it says it is, pretty sure it's the newest.

I am curious if it's just a graphical error in CF and truly has no relevance to anything being wrong on the board. I also have another brand new lux besides this one so I'll try installing it on that too.

The issue goes away when I roll back to BF 2.4 or CF 1.12.

Cheredanine, 2016-02-29:
Boris responded on the thread, he commented out a line and forgot to remove it, on rc4 acc,Bob and some other features do not work, even though the show up in configurator and appear to be enabled in CLI.
He will fix in rc5 but is working on other stuff he also wants in rc5

gruntwerkz, 2016-02-29:
Hey Cher nice to see you over here too.

So to confirm... The acc isn't working and won't work in rc4? Even thought it looks to be in CLI and sensor tab it is NOT working..... So no angle on rc4?

What is bob?

Thanks man, guess I'll have to wait to run this version.

Cheredanine, 2016-02-29:
Hi mate, yes confirmed, acc, blackbox (which my spell checker thinks should be called Bob) and I think a couple of other features will not work in rc4, for acc this is even if the icon is green, the feature is set and it is using CPU.

No angle, no horizon, no artificial horizon in multiwii

borisbstyle, 2016-03-01:
Fixed in the latest source

closed 2016-03-01 by borisbstyle