betaflight/betaflight #232: min_check not honored on 2.5.0-rc4

robertlacroix, 2016-02-29:
On my Naze32 it seems that min_check is not honored and it's impossible to arm by stick without breaking a prop or two. I didn't have the problem pre 2.5.0.

```
# dump

# version
# BetaFlight/NAZE 2.5.0 Feb 29 2016 / 01:04:11 (c050351)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature RX_PPM
feature VBAT
feature SOFTSERIAL
feature FAILSAFE
feature TELEMETRY
feature CURRENT_METER
feature ONESHOT125

# map
map TAER1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 30 4 115200 57600 0 115200
serial 31 0 115200 57600 0 115200

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
led 15 8,8:D:WF:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
led 19 10,12::R:3
led 20 9,13::R:3
led 21 8,14::R:3
led 22 7,14::R:3
led 23 6,13::R:3
led 24 5,12::R:3
led 25 5,11::R:3
led 26 6,10::R:3
led 27 7,9::R:3
led 28 0,0:::0
led 29 0,0:::0
led 30 0,0:::0
led 31 0,0:::0

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

# aux
aux 0 27 0 1425 2100
aux 1 28 0 1700 2100
aux 2 12 2 1575 2100
aux 3 0 0 900 900
aux 4 0 0 900 900
aux 5 0 0 900 900
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900

# adjrange
adjrange 0 0 0 900 900 0 0
adjrange 1 0 0 900 900 0 0
adjrange 2 0 0 900 900 0 0
adjrange 3 0 0 900 900 0 0
adjrange 4 0 0 900 900 0 0
adjrange 5 0 0 900 900 0 0
adjrange 6 0 0 900 900 0 0
adjrange 7 0 0 900 900 0 0
adjrange 8 0 0 900 900 0 0
adjrange 9 0 0 900 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0
adjrange 12 0 0 900 900 0 0
adjrange 13 0 0 900 900 0 0
adjrange 14 0 0 900 900 0 0

# rxrange
rxrange 0 1000 2000
rxrange 1 1000 2000
rxrange 2 1000 2000
rxrange 3 1000 2000

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 30
set rssi_ppm_invert = OFF
set input_filtering_mode = OFF
set rc_smoothing = OFF
set roll_yaw_cam_mix_degrees = 0
set max_aux_channels = 6
set debug_mode = NONE
set min_throttle = 1111
set max_throttle = 1850
set min_command = 1000
set servo_center_pulse = 1500
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 50
set use_oneshot42 = OFF
set use_multishot = OFF
set motor_pwm_rate = 1333
set servo_pwm_rate = 50
set disarm_kill_switch = ON
set auto_disarm_delay = 5
set small_angle = 25
set fixedwing_althold_dir = 1
set reboot_character = 82
set serialrx_provider = SPEK1024
set sbus_inversion = ON
set spektrum_sat_bind = 0
set telemetry_switch = OFF
set telemetry_inversion = ON
set frsky_default_lattitude =  0.000
set frsky_default_longitude =  0.000
set frsky_coordinates_format = 0
set frsky_unit = IMPERIAL
set frsky_vfas_precision = 0
set frsky_vfas_cell_voltage = OFF
set hott_alarm_sound_interval = 5
set battery_capacity = 2200
set vbat_scale = 110
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 33
set vbat_warning_cell_voltage = 35
set vbat_pid_compensation = OFF
set current_meter_scale = 174
set current_meter_offset = 60
set multiwii_current_meter_output = OFF
set current_meter_type = VIRTUAL
set align_gyro = DEFAULT
set align_acc = DEFAULT
set align_mag = DEFAULT
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 90
set max_angle_inclination = 700
set gyro_lpf = OFF
set gyro_sync_denom = 2
set gyro_soft_lpf =  60.000
set moron_threshold = 32
set imu_dcm_kp = 2500
set imu_dcm_ki = 0
set alt_hold_deadband = 40
set alt_hold_fast_change = ON
set deadband = 0
set yaw_deadband = 0
set throttle_correction_value = 0
set throttle_correction_angle = 800
set yaw_control_direction = 1
set yaw_motor_direction = 1
set yaw_jump_prevention_limit = 200
set tri_unarmed_servo = ON
set servo_lowpass_freq = 400
set servo_lowpass_enable = OFF
set acro_plus_factor = 30
set acro_plus_offset = 40
set failsafe_delay = 10
set failsafe_off_delay = 10
set failsafe_throttle = 1000
set failsafe_kill_switch = OFF
set failsafe_throttle_low_delay = 100
set failsafe_procedure = OFF
set rx_min_usec = 885
set rx_max_usec = 2115
set gimbal_mode = NORMAL
set acc_hardware = NONE
set acc_lpf_hz =  10.000
set accxy_deadband = 40
set accz_deadband = 40
set acc_unarmedcal = ON
set acc_trim_pitch = 0
set acc_trim_roll = 0
set baro_tab_size = 21
set baro_noise_lpf =  0.600
set baro_cf_vel =  0.985
set baro_cf_alt =  0.965
set baro_hardware = NONE
set mag_hardware = NONE
set mag_declination = 0
set pid_process_denom = 3
set blackbox_rate_num = 0
set blackbox_rate_denom = 0
set blackbox_device = SERIAL
set beeper_off_flags = 196
set magzero_x = 0
set magzero_y = 0
set magzero_z = 0

# rxfail
rxfail 0 a
rxfail 1 a
rxfail 2 a
rxfail 3 a
rxfail 4 h
rxfail 5 h
rxfail 6 h
rxfail 7 h
rxfail 8 h
rxfail 9 h
rxfail 10 h
rxfail 11 h
rxfail 12 h
rxfail 13 h
rxfail 14 h
rxfail 15 h
rxfail 16 h
rxfail 17 h

# dump profile

# profile
profile 0

set yaw_p_limit = 500
set pid_delta_method = MEASUREMENT
set dterm_lpf_hz =  0.000
set dterm_average_count = 4
set pid_controller = LUX
set p_pitch = 54
set i_pitch = 40
set d_pitch = 18
set p_roll = 42
set i_roll = 40
set d_roll = 13
set p_yaw = 90
set i_yaw = 50
set d_yaw = 0
set p_pitchf =  2.400
set i_pitchf =  0.400
set d_pitchf =  0.025
set p_rollf =  2.000
set i_rollf =  0.400
set d_rollf =  0.025
set p_yawf =  4.000
set i_yawf =  0.400
set d_yawf =  0.010
set level_horizon =  4.000
set level_angle =  4.000
set sensitivity_horizon = 75
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 30
set i_level = 30
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_expo = 70
set rc_yaw_expo = 70
set thr_mid = 50
set thr_expo = 0
set roll_rate = 75
set pitch_rate = 75
set yaw_rate = 75
set tpa_rate = 0
set tpa_breakpoint = 1500

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_expo = 70
set rc_yaw_expo = 70
set thr_mid = 50
set thr_expo = 0
set roll_rate = 75
set pitch_rate = 75
set yaw_rate = 75
set tpa_rate = 0
set tpa_breakpoint = 1500
```

robertlacroix, 2016-02-29:
Maybe it's time to start using switch arming :)?

Yamaford, 2016-02-29:
your min check is 1100! how are you even arming?

Yamaford, 2016-02-29:
Hmm I switch arm but I don't think that is the solution. Airmode doesn't kick in until you go above mincheck the first time. If you switch arm, Yaw (normally) is disabled when you go below min check no matter airmode or not.

borisbstyle, 2016-03-01:
fixed

closed 2016-03-01 by borisbstyle