betaflight/betaflight #233: Yawing when arming and disarming under 2.5

Bozbi, 2016-02-29:
When i arm or disarm, my quad yaws left or right. 
Seems to be a min_command, min_check, min_throttle issue.
How can i fix this?

RC-Slater, 2016-02-29:
Use a switch to arm and disarm.

Bozbi, 2016-02-29:
That's the only solution?

RC-Slater, 2016-02-29:
Yea why not? It completely eliminates the possibility of a wild yaw on takeoff or landing. Boris, josh bardwell, and many others recommend switch arming. 

If you insist, you can try increasing min_check by 20 points or so.

deepimpactio, 2016-02-29:
@Bozbi if you don't have motor stop feature and/or Airmode is active this the normal behavior. You would have to use switch arm instead of stick arm

robertlacroix, 2016-02-29:
Probably a dupe #232

Yamaford, 2016-02-29:
Hmm I switch arm but I don't think that is the solution. Airmode doesn't kick in until you go above mincheck the first time.  If you switch arm, Yaw (normally) is disabled when you go below min check no matter airmode or not.

deepimpactio, 2016-02-29:
@Yamaford yes now that you mention it that's right.

borisbstyle, 2016-03-01:
This is fixed in the latest source. Stick armers will get some deadband below min_check

closed 2016-03-01 by borisbstyle