betaflight/betaflight #239: Increase Accelerometer Range/scale to 16G

goatspoon, 2016-03-02:
The accelerometers (MPU6050) are currently set to use a full scale range of +-8G. This range is being saturated in full throttle turns and punches.

borisbstyle, 2016-03-02:
If someone has the time to look into it it would be nice. There are many different drivers which all handle it differently.
Are you aware any other opensource projects using it?

jrwrodgers, 2016-03-02:
Code:
/** Power on and prepare for general usage.
 * This will activate the device and take it out of sleep mode (which must be done
 * after start-up). This function also sets both the accelerometer and the gyroscope
 * to their most sensitive settings, namely +/- 2g and +/- 250 degrees/sec, and sets
 * the clock source to use the X Gyro for reference, which is slightly better than
 * the default internal clock source.
 
void MPU6050::initialize() {
    setClockSource(MPU6050_CLOCK_PLL_XGYRO);
    setFullScaleGyroRange(MPU6050_GYRO_FS_250);
    setFullScaleAccelRange(MPU6050_ACCEL_FS_**3**);
    setSleepEnabled(false);
}

Where:
 AFS_SEL | Full Scale Range | LSB Sensitivity
 --------+------------------+----------------
 0       | +/- 2g           | 8192 LSB/mg
 1       | +/- 4g           | 4096 LSB/mg
 2       | +/- 8g           | 2048 LSB/mg
 3       | +/- 16g          | 1024 LSB/mg

Reference:
http://www.i2cdevlib.com/docs/html/class_m_p_u6050.html#abd8fc6c18adf158011118fbccc7e7054

Drivers:
https://github.com/jrowberg/i2cdevlib/tree/master/Arduino/MPU6050

Does this help?

borisbstyle, 2016-03-02:
Ok seems easy

Can someone test/review this PR I quickly coded
https://github.com/borisbstyle/betaflight/pull/241

goatspoon, 2016-03-03:
Those changes and the same that I made to test and they worked. Now I just need to figure out how to saturate 16G.

borisbstyle, 2016-03-03:
@goatspoon 
So you did test it? can you confirm if it works with flight tests

goatspoon, 2016-03-03:
It works fine on my setup which is the MPU6050 on the SPRacingF3. The raw accel is half what it normally is. Note: I have not tested any kind of angle/horizon mode. The blackbox is correct but blackbox explorer converts the raw value incorrectly to 0.5G when it should be 1G.

The change is good for me though, I can now get accel values above 8G comfortably.

borisbstyle, 2016-03-03:
I am interrested in flight tests to see if the pid controllers will use the right values. Blackbox stuff can be fixed though

borisbstyle, 2016-03-03:
is this table correct? As the existing code was sete to  4096 and not 2048. That would mean only 4G

Where:
AFS_SEL | Full Scale Range | LSB Sensitivity
--------+------------------+----------------
0 | +/- 2g | 8192 LSB/mg
1 | +/- 4g | 4096 LSB/mg
2 | +/- 8g | 2048 LSB/mg
3 | +/- 16g | 1024 LSB/mg

borisbstyle, 2016-03-03:
I would say the 16G would need to be int16_t / 16 and for 8G int16_t / 8.

That would correspond with 
8G scale 1G= 4096
16G scale 1G = 2048

borisbstyle, 2016-03-03:
acc_1G = 512 * 8 would have to become acc_1G = 512 * 4

jrwrodgers, 2016-03-03:
I can't confirm if that table is correct ...it was from the i2cdevlib mpu6050 class functions.. Sorry I'm still trying to learn

borisbstyle, 2016-03-03:
@goatspoon 
Can you retry my latest change in PR. Basically changed multiplier from 2 to 4. Blackbox should also give proper numbers as it uses real scaling

MasterZap, 2016-03-03:
But does this matter? BAck when I was researching ACC behaviour, it turned out that any Acc vector that is not (very near to) 1G is just thrown away... so only if the vector is near 1g is it even used to update the estimate of what is "down".

Now this was a few months ago, and with Boris speed of changing things, my understanding may be COMPLETELY outdated :)

MasterZap, 2016-03-03:
...and yes, now I checked the code, this HAS changed completely. :)

But wasn't there some sense to the old logic? Basically, the old code was maintaining an estimated attitude (an estimated down vector) that was updated (rotated) by the gyros.

It also took an accelerometer measurement. If the acceleromter measurement was near 1G in length, it gently nudged the estimated vector in the direction of the measured value (very slowly, I think the default was 1/1000:th of the distance per loop) to behave as a kind of filter....

The current code does none of that, that I can see?

/Z

borisbstyle, 2016-03-03:
@MasterZap 
The new IMU logic is based on quaternion.....@digitalentity did a full rework of that.

MasterZap, 2016-03-03:
Right... but the code that ignores large acc vectors is gone. I feel that is wrong. A non-unity acc vector is by definition not gravity and should not update the attitude estimate!!

borisbstyle, 2016-03-03:
@MasterZap 
Can you suggest a PR with adding the vectors back so it can be tested and compared?

In the meanwhile 16G will be tested and see if the reduced resolution could impact anything. I dont expect so.

closed 2016-03-03 by borisbstyle

MasterZap, 2016-03-14:
I see if I can whip something up in my copious free time. (of which I have none, and if I have, I tend to fly... but we'll see... bigger miracles has occured) :)

/Z

digitalentity, 2016-03-14:
@MasterZap 

> Right... but the code that ignores large acc vectors is gone. I feel that is wrong. A non-unity acc vector is by definition not gravity and should not update the attitude estimate!!

Code is there. Anything outside of range 0.90g - 1.10g is ignored. See: 
https://github.com/borisbstyle/betaflight/blob/betaflight/src/main/flight/imu.c#L356
https://github.com/borisbstyle/betaflight/blob/betaflight/src/main/flight/imu.c#L388-L390

MasterZap, 2016-03-14:
Oh... you are so right. I assumed that was an error checking function and didn't look into it. Mea Culpa.

HOWEVER.... I always thought the acc should be mixed in weighted to how close it is to a unit vector, not this hard cutoff. I imagine that would be smoother.

Secondly, also... if we only care about 0.90g - 1.10g, why are we increasing scale to 16G? Please explaininate :)

borisbstyle, 2016-03-14:
@MasterZap I increased it to 16G. Perhaps the threshold of 0.9-1.10G needs to be increased too to have more accuracy after those flips and rolls?

MasterZap, 2016-03-15:
First, I'd like to hear some motivation... ANY motivation... for the 16G increase. It makes no sense to me.

Secondly, yes, 0.9-1.1 seems like a very narrow band, and quite arbitrary (and hasn't changed since the oldest code I can recall of MultiWii).

I'd experiment with things like:
- wider band?
- weighted effect? (I.e. the closer it is to 1.0, weight it more)

Heck, if you wanted to be really nerdy you could take known motor thrust out and subtract that acceleration vector to get the "real" gravity vector back.  Or something.

/Z