betaflight/betaflight #240: Compilation error need some help

gamani, 2016-03-02:
Just merged and tried to recompile SPRACINGF3 and got the following errors. What did i miss ?

Thanks guys

`C:\cygwin64\tmp\ccqAEiqn.ltrans0.ltrans.o: In function `ak8963Detect':
C:\cygwin64\home\xx\betaflight/./src/main/drivers/compass_ak8963.c:139: undefined reference to `mpu6500WriteRegister'
C:\cygwin64\home\Alxxex\betaflight/./src/main/drivers/compass_ak8963.c:142: undefined reference to `mpu6500WriteRegister'
C:\cygwin64\home\xx\betaflight/./src/main/drivers/compass_ak8963.c:145: undefined reference to `mpu6500WriteRegister'
C:\cygwin64\tmp\ccqAEiqn.ltrans11.ltrans.o: In function `ak8963SPIWrite':
C:\cygwin64\home\xx\betaflight/./src/main/drivers/compass_ak8963.c:109: undefined reference to `mpu6500WriteRegister'
C:\cygwin64\home\xx\betaflight/./src/main/drivers/compass_ak8963.c:110: undefined reference to `mpu6500WriteRegister'
C:\cygwin64\tmp\ccqAEiqn.ltrans11.ltrans.o:C:\cygwin64\home\xx\betaflight/./src/main/drivers/compass_ak8963.c:111: more undefined references to `mpu6500WriteRegister' follow
C:\cygwin64\tmp\ccqAEiqn.ltrans11.ltrans.o: In function `ak8963SPIRead':
C:\cygwin64\home\xx\betaflight/./src/main/drivers/compass_ak8963.c:102: undefined reference to `mpu6500ReadRegister'
collect2.exe: error: ld returned 1 exit status
`

borisbstyle, 2016-03-02:
Sounds like you are missing a lot of stuff

gamani, 2016-03-03:
MMm, i've just updated the target.h from the SPRACINGF3 folder with the following.

`#define GYRO
#define USE_GYRO_MPU6500
#define GYRO_MPU6500_ALIGN CW270_DEG

#define ACC
#define USE_ACC_MPU6500
#define ACC_MPU6500_ALIGN CW270_DEG

#define BARO
#define USE_BARO_MS5611
#define USE_BARO_BMP280

#define MAG
#define USE_MAG_AK8963
//#define USE_MAG_HMC5883
//#define MAG_AK8963_ALIGN CW270_DEG`

borisbstyle, 2016-03-03:
re-fork it and it should be fine.

gamani, 2016-03-03:
OK forked and updated/added the following files

accgyro_spi_mpu9250.c
accgyro_spi_mpu9250.h
compass_ak8963.c with > verifympu9250WriteRegister and mpu9250ReadRegister function instead of the mpu6500WriteRegister and mpu6500ReadRegister one.

Changed the target.h in spracingf3 folder and makefile like upper.

And got the following errors

C:\cygwin64\tmp\cc1qh2q9.ltrans0.ltrans.o: In function `ak8963Detect':
C:\cygwin64\home\Alex\betaflight/./src/main/drivers/compass_ak8963.c:140: undefined reference to `verifympu9250WriteRegister'
C:\cygwin64\home\Alex\betaflight/./src/main/drivers/compass_ak8963.c:143: undefined reference to `verifympu9250WriteRegister'
C:\cygwin64\home\Alex\betaflight/./src/main/drivers/compass_ak8963.c:146: undefined reference to `verifympu9250WriteRegister'
C:\cygwin64\tmp\cc1qh2q9.ltrans11.ltrans.o: In function `ak8963SPIWrite':
C:\cygwin64\home\Alex\betaflight/./src/main/drivers/compass_ak8963.c:110: undefined reference to `verifympu9250WriteRegister'
C:\cygwin64\home\Alex\betaflight/./src/main/drivers/compass_ak8963.c:111: undefined reference to `verifympu9250WriteRegister'
C:\cygwin64\tmp\cc1qh2q9.ltrans11.ltrans.o:C:\cygwin64\home\Alex\betaflight/./src/main/drivers/compass_ak8963.c:112: more undefined references to `verifympu9250WriteRegister' follow
C:\cygwin64\tmp\cc1qh2q9.ltrans11.ltrans.o: In function `ak8963SPIRead':
C:\cygwin64\home\Alex\betaflight/./src/main/drivers/compass_ak8963.c:103: undefined reference to `mpu9250ReadRegister'
collect2.exe: error: ld returned 1 exit status
Makefile:856 : la recette pour la cible « obj/main/betaflight_SPRACINGF3.elf » a échouée
make: *** [obj/main/betaflight_SPRACINGF3.elf] Erreur 1

Btw i'm trying to make the firmware compatible with an SPRACING F3 with OSD integrated. Don't know if you can add the support yourself ?

Thanks for help

gamani, 2016-03-12:
Hello Boris, i'm still stuck on this problem. Can you help ? I think you can solve it better than me :)

closed 2016-04-04 by borisbstyle