betaflight/betaflight #244: SPR F3 Mini not recognising Radio....

BoosiveFPV, 2016-03-03:
Hi Boris, Just flashed my mini with this firmware. But for some reason it won't recognise my TX :/. I flashed it back to Cleanflight and everything worked fine. So it's definitely a firmware issue. Hopefully this issue can be fixed. I'll try some of your older firmwares to see if they work.

RC-Slater, 2016-03-03:
Nope, mine works just fine on 2.5.0.  Definitely not a firmware issue.

BoosiveFPV, 2016-03-03:
How come it doesn't happen on 2.4.1 then?  😕

RC-Slater, 2016-03-03:
Make a video showing your RX/TX working on 2.4.1, flashing 2.5.0, restoring your settings and it not working.  Need CLI dump... or some more info at least.

BoosiveFPV, 2016-03-03:
I'm using a D4R. This is doing my nut in tbh. My naze32 was plug and play pretty much.

BoosiveFPV, 2016-03-03:
Is PPM actually enabled on this code?

borisbstyle, 2016-03-04:
Continue discussion here
https://github.com/borisbstyle/betaflight/issues/247

closed 2016-03-04 by borisbstyle