betaflight/betaflight #247: No PPM for SPRF3 Mini

BoosiveFPV, 2016-03-04:
There is no PPM support for the mini. It is like its not been enabled for this code. It works on 2.4.1 but not 2.4.0 or 2.5.0. Everybody that has it working that I've seen are using SBUS. Please can you fix this asap.

borisbstyle, 2016-03-04:
@BoosiveFPV 
If it works on 2.4.1 it should also work on 2.5.0. Nothing has been changed there actually. Are you sure it doesnt work?

BoosiveFPV, 2016-03-04:
I know it should. I've not changed anything apart from flash the firmware. Reflashed countless amounts of times but its like PPM is invisible. Something elsewhere you changed might have caused an issue with PPM. Have you got a mini to test or know anyone that has?

borisbstyle, 2016-03-04:
Try disabling oneshot

BoosiveFPV, 2016-03-04:
It works! :D. Cheers boris you legend. Now how do I enable multishot lol

borisbstyle, 2016-03-07:
Can you also check 2.5.3 with PPM and oneshot. Open a new issue if it still exists

closed 2016-03-07 by borisbstyle

BoosiveFPV, 2016-03-07:
Yeh I will try it tomorrow when I sort my quad out. Will let you know.