betaflight/betaflight #249: Quad tilting after punches

jsweeney91, 2016-03-04:
https://www.youtube.com/watch?time_continue=12&v=RQ9VXFBuMnk

laurynas-karvelis, 2016-03-04:
+1, I have similar, but much more pronounced issue with 2.5.0 final version. Could it be accelerometer related issue with widening the range from 8G to 16G? I'm flying on rate mode btw.

le3dmax, 2016-03-04:
I was getting that awhile ago when I was flying with  PPM rx.

lefticals, 2016-03-04:
Same issue after a punchout when i cut throttle and roll seems like quad has a mind of its own for s plit second ... Either yaw or roll moves a few degrees

borisbstyle, 2016-03-04:
Looks like failsafe. The quad wants to level in those glitches. Put a beeper when in failsafe and you will know that

laurynas-karvelis, 2016-03-04:
Hmm, weird... Never had such behaviour prior to upgrading to 2.5.0 final. SP3

borisbstyle, 2016-03-04:
I actually flew 20 packs yesterday on 2.5 and spf3 with SBUS....not a
single.issue

jsweeney91, 2016-03-04:
This is using SBUS btw on naze32 rev6 with . Will do @borisbstyle but why would failsafe be engaged when the quad is right next to me on two different rx?

jsweeney91, 2016-03-04:
softserial too

carebear482, 2016-03-04:
I can confirm this issue with naze rev5 SBUS.

dawnduran, 2016-03-04:
Confirm Issue with naze rev5 CPPM (D4R-II)

borisbstyle, 2016-03-04:
increase pids perhaps....it happens on punch outs. So could be tuned out i
guess

dawnduran, 2016-03-04:
Increase PIDs is not the solution. Quad starts wobbeling then. It doesn't ohnly happens on punchouts. It happens too, when you do many rolls an flips.

le3dmax, 2016-03-04:
Are you using Airmode? That happens with me when I forget to use air mode
with motor stop on.

borisbstyle, 2016-03-04:
In acro mode or horizon?

le3dmax, 2016-03-04:
I fly acro.

dawnduran, 2016-03-04:
acro, no - airmode on or off - issue is the same.

husyildiz, 2016-03-04:
https://github.com/cleanflight/cleanflight/issues/791#issuecomment-108163562

When I was using my naze32 10dof with cleanflight it drifts and tilts one
side when full throttle.

I have able to solve issue above links describes comment.

Maybe helpfull. Sorry if it doesn't helps.

husyildiz, 2016-03-04:
@hydra : the outcome of all of this is...

    acc_lpf_factor should default to 0
    gyro_cmpf_factor should be 1000

With these settings a quad with noisy acc data flies better than with other
settings, and it has no adverse effect otherwise.

The other benefit of acc_lpf_factor=0 is that in Configurator the user will
see the actual vibration of the quad, affected only by gyro_lpf, not a
smoothed version.

This comment above.
4 Mar 2016 22:18 tarihinde "hüseyin yıldız" <husyildiz@gmail.com> yazdı:

borisbstyle, 2016-03-04:
on punches quad loses power and the weak side will drift if the quad is not
balanced.
When something only happens on punches it really doesnt sound like a
firmware issue anyway

le3dmax, 2016-03-04:
I remember another thing, something like that happened once with me when I
was using a futaba with Frsky DIY module.

husyildiz, 2016-03-04:
Above comment doesn't means firmware issue. It says more vibration causes
this . So soft filter resolved it with cleanflight. Only suggestion . I
thought it may help.

Schakus, 2016-03-04:
hey boris
please look at my video https://youtu.be/T2x_CKWRMos
i think thats the same bug. both crashes with 2.5.0

micaheli, 2016-03-04:
I've had this same issue with other firmwares (cleanflight) as well as
betaflight on one of my quads.  This is likely due to mechanical or
electrical failure or defects.  One of my ESC's doesn't perform quite as
well as it used to and two motors have been cooked pretty good but still
work.  I've also seen this in quads with improperly calibrated ESC
endpoints where one side was maxing out and the other wasn't reaching WOT.

dawnduran, 2016-03-04:
I go back to Betaflight 2.3.1 where this Problem doesnt exist.

Schakus, 2016-03-04:
@micaheli i have this error since 2.5.0 but not every battery. i flyed 3 batterys, all ok but the next battery i have this bug...

borisbstyle, 2016-03-04:
also when battery power becomes too low the quad.might nit hold its angle

dawnduran, 2016-03-04:
https://youtu.be/BJ5qxgwGe8w

borisbstyle, 2016-03-04:
@dawnduran 
That's not tilting after punches. Looks like yaw dominates the roll there. Try 2.5 version and see if it makes difference for your setup

RC-Slater, 2016-03-04:
There's some discussion of a mysterious I-term dropout in the rcgroups thread. Any relation?

dawnduran, 2016-03-04:
Mensch Boris. Ich habe nur ein Video von der 2.4.1. Bei deiner Version 2.5 ist das Problem genau das gleiche. Nur bei der Version 2.3.1 und darunter existiert das Problem nicht.

borisbstyle, 2016-03-04:
@RC-Slater
Since when is Iterm reset below min_check misterious (without airmode)?

SoulPictures, 2016-03-04:
anyone with this problem, who has a blackbox installed and could shed some light on this ?

Schakus, 2016-03-04:
i have a blackbox, but i didnt record the flight :-( i will do it on sunday. 
do you see my video boris? what do you mean?

dawnduran, 2016-03-05:
Boris, I only have an video of testing 2.4.1. But on 2.5.1 its the same. Only 2.3.1 and before the problem does NOT exist!!! 

I gone back to 2.3.1 and it flys perfect!

It's NOT a "increase-P" problem!
It's NOT a "better Battery" problem!
It's NOT a "too much yaw" problem!
It's NOT a "update Betaflight" problem!

on some short hard punchouts, it nicks slighty and short to the left side.
on some longer hard punchouts, it roll maybe 150° over left and 1s you can do what you want on your Taranis, no inputs are working.
on some rolls or flips (some rolls and flips before), that rolls hard over left (Video from schakus).

Fully setup:
Alien 5"
Naze 32 Rev. 5
Kiss 24A
T-Motor F40 2500kv
Graphene 1300mah 4S 65C
HQ 5x4,5x3BN; HQ 5x4x4
FrSky D4R-II CPPM
Weight: 550gr
Telemetry Inversion ON (Softserial)
2 AUX-Channels used
2khz
Airmode ON/OFF (same problems)
Roll: P4.1; I35; D13
Pitch: P4.5; I40: D18
RC Rate: 0.90
Roll Rate 0.85
Pitch Rate 0.80

Hope you can check this. Maybe it's the ITerm-Dropout-Problem?!

borisbstyle, 2016-03-05:
there is no iterm drop in the current version.

Did you calibrate your escs? Something must be wrong on that setup ad you
are the only one having this on punches

carebear482, 2016-03-05:
He's not the only one. Happened to me 4 times in two hours of flying.

Schakus, 2016-03-05:
And what is with me? :-(

borisbstyle, 2016-03-05:
1 guy posted a vid where his quad twitched during hard turn.....one says
that the quad tilts during punches. Are we talking about same issue?

borisbstyle, 2016-03-05:
And its a long while from 2.3.1.....i really didnt see anybody else
reporting this on 2.3.3  2.3.4....2.4.0...2.4.1...

dawnduran, 2016-03-05:
Boris. Again:

THE QUAD FLYS ABSOLUTE PERFECTLY ON 2.3.1.
Calibration of ESC is not needed...

dawnduran, 2016-03-05:
I switched from 2.3.1 to 2.4.1 - problem is new, but not often.
i switched to 2.5.0 RC7 - problem still exists, very often
i switched to 2.5.0 final - problem still exists, very often
i switched to 2.5.1 - problem still exists, very often
i switched back to 2.3.1 final - problem gone

Schakus, 2016-03-05:
Boris my problem is since 2.5, before i didnt have it. But it is not reproducible. I think its the same problem

borisbstyle, 2016-03-05:
The other guy has it on every version except 2.3.1 :)

It is confusing having different issues.

But lets see....2.3.1 was going down on power when 1 motor was reaching
full power. The other versions (except 2.3.5) will just give you full power.
This means that when you punch there is a problem on one of your motors
maxing out and mismatching others why it should drift on one side.
So in a certain way 2.3.1 is masking the real problem.
Hook up a blackbox and find the motor/esc causing issues.

SoulPictures, 2016-03-05:
Dawn you got an Naze Rev5 ? got some Winbond SPI flash chips left to upgrade a Blackbox

dawnduran, 2016-03-05:
sounds plausible Boris. I will change the left front motor.

borisbstyle, 2016-03-05:
well check the esc too as that one is mostly the problem. Like when one of
the fets would die

jsweeney91, 2016-03-05:
okay I flew with failsafe set to hold and now im not getting the twitches but failsafe is being engaged, I'm checking my blackbox recording just now to see if I can locate the issue. Will upload it with video later if I got the right blackbox data

jsweeney91, 2016-03-05:
> there is no iterm drop in the current version.
> 
> Did you calibrate your escs? Something must be wrong on that setup ad you
> are the only one having this on punches

@borisbstyle I'm experiencing it on punches too, not on hard turns

jsweeney91, 2016-03-06:
issue seems to be fixed after disabling motor stop and auto mode with fail safe. Probably just masking an actual issue with the failsafe though

borisbstyle, 2016-03-07:
I am going to close this issue. If anyone has issues on 2.5.3 let me know.

For those with failsafe issues.....please recheck your receivers etc.

closed 2016-03-07 by borisbstyle