betaflight/betaflight #251: ANGLE an HORIZON modeas are missing in 2.5.2

skaman82, 2016-03-05:
Is that intentional? Can't believe that.

borisbstyle, 2016-03-05:
they are not missing. I think your accelerometer is disabled :)

borisbstyle, 2016-03-05:
when going low looptimes acc gets disabled automatically

skaman82, 2016-03-05:
Just flashed my spracing F3, din't played with looptimes. So on default the leveled flightmodes are missing. How can I get them back?

Highoctane32, 2016-03-05:
CLI 
set acc_hardware = AUTO

borisbstyle, 2016-03-05:
no there are not missing on default.

Your settings are kept the same....not reset from previous release

Highoctane32, 2016-03-05:
Agree BTW Boris I just noticed if I set looptime to 1000 via config page on naze32 target(checking default sensor settings) acc and Baro come on as expected but loop time is actually set to 125!

borisbstyle, 2016-03-05:
redownload the latest binaries of 2.5.2.....auto config wasnt working properly

closed 2016-03-05 by borisbstyle