betaflight/betaflight #252: motor commands reversing at high/low RC input WTF?

CreeT6, 2016-03-06:
when i put my rudder stick at around 1230us the rudder output will reverse! it will suddenly go from hard right to hard left! this only happens to the rudder controls for what ever reason. my RC channel range is set to 1000 - 1520 - 2000, SBUS. Rewrite default PIDs. CC3D 2.5.2 4khz [but looptimes are around 265]. i will post a !dump if no one else is dealing this

borisbstyle, 2016-03-06:
are you sure this is not your rx issue?

CreeT6, 2016-03-06:
@borisbstyle pretty sure not, i just flash 2.4.1 and everything is fine just like 1 week ago

borisbstyle, 2016-03-06:
i do full yaws all the time without isssues during testing :)
It wad the main test scenario yesterday actually.

Something must be wrong on your setup. Can you check rx tab?

CreeT6, 2016-03-06:
@borisbstyle rx tab is normal, btw i was doing the testing without props, stationary. it getting late 11:30pm, gona to flying tests tmr. good night

borisbstyle, 2016-03-07:
I am going to close the issue as I could not replicate this behaviour and nobody else reported it

closed 2016-03-07 by borisbstyle

jsweeney91, 2016-03-24:
@CreeT6 can you get a video of the behavior? Sounds like you might have a similar issue as me