betaflight/betaflight #254: Choosing LUX makes board not arm

cmavr8, 2016-03-06:
On a fresh BF 2.5.2 install, with minimal settings changed, on a naze32 rev5, I'm seeing this behavior:
In PID tuning tab of the configurator, when I switch from MWREWRITE to LUX, I can no longer arm the board.

The only two settings that seem to change (according to dump + diff) are:
`set gyro_sync_denom = 8
set pid_controller = MWREWRITE`

After changing to LUX, I get these set:
`set gyro_sync_denom = 1
set pid_controller = LUX`

Although in the GUI, I see the loop time also being dropped.

borisbstyle, 2016-03-06:
change manually to gyro_sync_denom = 8

Will be fixed in the next release.

borisbstyle, 2016-03-06:
just tested and it actually works fine.

Changing rewrite to lux gives gyro_sync_denom = 8

EduardTheThird, 2016-03-07:
I have experienced the same arming issues on SP F3 Deluxe using BF 2.5.2 clean install at 2k looptime. Thanks for the amazing work @borisbstyle . Will try gyro_sync_denom = 8

borisbstyle, 2016-03-07:
Going to close this issue. 

Read wiki about CPU usage and monitoring. Thanks.

closed 2016-03-07 by borisbstyle

cmavr8, 2016-03-08:
Thanks boris, you're right, I hadn't read that.

If you think it's ok for gyro_sync_denom to change to an "unworkable" value, when the user changes to LUX, ok. Otherwise you may consider not switching to lower gyro sync values automatically, and letting the user dial the value in manually. 

I don't have enough knowledge to have an opinion. I'll happily work around it anyways. :) Thanks for all the good work!