betaflight/betaflight #255: Some modification of led-strip functionality

maxsymba, 2016-03-06:
Hi.

I did some modification in "led-strip" for INDICATOR and RING processing.
https://github.com/maxsymba/betaflight/blob/betaflight/src/main/io/ledstrip.c
Now INDICATOR works only on LED's with defined directions and have different colors:
WEST - turn left, orange
EAST -  turn right, orange
NORTH - acceleration, blue
SOUTH - decceleration, red
RING: phase now depends on LED's position on the coordinate grid. Animation direction also now depends on position. If (X+Y) increases - this is a one direction, if X+Y decreases - another direction.
Please test it. Perhaps you will like this new tuning.
In general I think that there is a sense to seriously modify this module.

Best regards from Ukraine,
Max

fahrvergnuugen, 2016-03-25:
It would be cool to have some new modes to allow customized animations (like the thrust ring), or strobing. It would probably be quite a bit of work on the UI side though...

maxsymba, 2016-03-25:
Can you show an example?

KiteAnton, 2016-06-23:
Related to #400

closed 2016-06-23 by KiteAnton