betaflight/betaflight #256: TBS Powercube current sensor?

DaTwo, 2016-03-07:
I see that BST is implemented in betaflight. Is the current sensor on the TBS Powercube supported? I can't get it working if so. Love some help getting this working if it is indeed supported. I am trying to output current (mah used) to a MWOSD for my mini quad.

Thanks

freddiebd, 2016-03-13:
Working for me with Powercube (2.5.3) and Core Pro

DaTwo, 2016-03-13:
Are you sure its measuring current? I checked the pin on the cpu itself and its going to a resistor divider. I am able to see the value changing as I alter the voltage on the pin. This tells me that the reading is from the ADC (I suspect its the 5 volt rail) not the current sensor via i2c. If there is already a way to make it read the sensor via i2c I would love to know how you did it. Thanks.

DaTwo, 2016-03-13:
The TBS Powercube uses a ina230 current monitor to measure current. Its connected to the i2c bus via the BST pins on the stacking header.

I am more than willing to test but unfortunately I am more of a hardware guy than a software guy for this task.

Here is the datasheet.
http://www.ti.com/lit/ds/symlink/ina230.pdf

freddiebd, 2016-03-14:
You sir are correct, It appears that with 2.5.3 the BST has broken- went back to my previous version 2.4.1, (I think), and it works :+1:

freddiebd, 2016-03-14:
@larryho5

DaTwo, 2016-03-14:
Wow, so 2.4.1 works on the current for sure? Great news. I am on 2.5.3. Will have to give it a shot

DaTwo, 2016-03-14:
Freddiebd, is cleanflight reporting this as current as shown here?

[image]

larryho5, 2016-03-15:
Sorry, the current sensor on the PowerCube not reporting to CleanFlight now, it is report to CorePro. We may catch this message on the later version.

DaTwo, 2016-03-16:
@larryho5 , is the i2c data just a pass through to the CorePro as of now? If this was implemented the cube would be my goto for all my builds. I really hope this gets picked up and implemented. It makes the cube a perfect solution for many this way.

DaTwo, 2016-05-19:
@larryho5 , any chance this can be looked into yet?

larryho5, 2016-05-26:
@DaTwo , I think it should be possible, please wait for a while.

jm-wijnholds, 2016-06-13:
@borisbstyle Any idea with version of BetaFlight will support BST again? would be great to use the current sensor of my powercube for my minimOSD :).

borisbstyle, 2016-06-13:
Doesnt it all work in 2.8.1?

DaTwo, 2016-06-13:
As far as I can tell no. Still no way for betaflight to read the current sensor on the cube.

borisbstyle, 2016-06-13:
You need to report that to TBS and they will probably fix it soon

larryho5, 2016-06-13:
@DaTwo , Sorry, i will do that later, just don't have time yet.

martinbudden, 2016-07-06:
@larryho5 , I've marked this for the v3.1 release.

borisbstyle, 2016-07-09:
Oops

borisbstyle, 2016-07-27:
@larryho-5
Let me know if you want to fix this before the next release

DaTwo, 2016-07-27:
@larryho5 , Please please please get this working.  :)

larryho5, 2016-07-28:
@DaTwo and @borisbstyle  , Sorry, I have been too busy, hopefully, I will have time to take a look on some day next week.

DaTwo, 2016-08-01:
@larryho5 , any luck? Really hope to see this working in the next release.  Thanks again.

durma, 2016-08-08:
is there any progress on this? @borisbstyle  please please solve this as I really would need to know the amperages.

borisbstyle, 2016-08-09:
Tbs guys need to solve this ;)

In betaflight everyone is responsible for their own hardware

DaTwo, 2016-08-09:
Sounds like its time to sell this thing(s). Without a working current meter I have come way to close to killing a few batts by running too low. I don't think TBS has any interest in making this work. Larryho5 was my hope and it sounds like he is too busy at this time to finish it.

borisbstyle, 2016-08-09:
but also try 3.0 as well to verify that it doesnt work there too

DaTwo, 2016-08-09:
@borisbstyle , As far as I can tell nothing has been done on this issue. 3.0 still is not working. The hardware is all there hooked up to I2C on the FC. Its just a matter of reading it and setting the FC current to its output. Its a shame since its a software issue only. Hopefully someone will pickup the torch and make this work.

larryho5, 2016-08-09:
@DaTwo @borisbstyle , sorry guys, too busy now.
you may skype me at "larryho5@outlook.com"

martinbudden, 2016-11-01:
@larryho5 , any chance of getting this fixed for the 3.1 release?

martinbudden, 2016-12-02:
Deferred to 3.2, since no response from @larryho5

larryho5, 2016-12-03:
@martinbudden How you want to voltage send? by BST?

martinbudden, 2016-12-03:
It's been a while and I can't remember the details of this problem. I'm also fairly busy over the next few days, so can you ask @ borisbstyle

mdrjr, 2016-12-17:
@larryho5 It would be nice to know how many amps is being pulled. Since this information is already present on the BST port, that's what we are asking to have it on the FC as well.

This will enable for instance, current and consumed mA/h on other OSD's and for FRSky SmartPort telemetry.

mdrjr, 2017-03-09:
TBS Support is amazing! 1 year this issue....

geins, 2017-07-03:
As this is also an issue for me, mainly due to trying get this to work for telemetry on my radio, I contacted the Helpdesk. Here is the undesirable answer:

...
The power cube is designed to be used with the CORE PRO or the FPVision OSD. This why there is no implementation inside betaflight for reading the current sensor data directly. If there is somebody from the betaflight community who wants to write this code we are happy to share the specifications. 

Thank you for your understanding. 
...

Well, too bad

mikeller, 2017-07-03:
@geins: I guess that is quite all right, they provider firmware for their hardware, and nobody can force them to spend more effort on supporting alternative firmwares. If they are willing to share the specification, and can be convinced to throw in some engineering samples of the hardware, it is well possible that a developer will volunteer to add support for their current sensor to Betaflight.

Can you ask them to join the Betaflight slack (registration at http://www.betaflight.ch/) to discuss this if they are interested?

geins, 2017-07-04:
@mikeller sure I can ask them :)

Well, I know many people not using an OSD, hence you really don't need one when u have a telemetry radio, so you can skip that wheight. 
In my opinion, betaflight is not an alternative firmware, since it's their first choice, delivering it from factory. They sell the powercube separately, so they should fully support it's features. It makes no sense to me that the current sensor is not supported. 
I personally fly the TBS Vendetta, having the drawn current in the OSD from the Core Pro. But being in flying combat, you can fry your batteries, overseeing the current. When my radio rings an alarm, everything would be clear :)
I guess my next build will be by KISS...

closed 2022-05-08 by haslinghuis