betaflight/betaflight #257: SPRacingF3 will not arm 2.5.3 d4r-II

slacker87, 2016-03-08:
SPRacingF3 will not arm. Receiver showing ok, switch shows as fine. Disabling failsafe does not help arm nor does disabling vbat monitoring. Signal and arm was fine on cleanflight and older betaflight versions. Motors, all other switches, and LED strip all showing normal receiver info, as well as telemetry.

slacker87, 2016-03-08:
Cleanflight 1.12.0 same exact settings 1 to 1 will arm every time.

borisbstyle, 2016-03-08:
check on wiki what to do.

check cpu, cycletime etc

slacker87, 2016-03-08:
Should it not just work "out of box"?

borisbstyle, 2016-03-08:
Yes it should and it works out of box. It always depends of what features you have enabled.
I often see folks using luxfloat, softserial, ledstrip telemetry on F1 boards and many things and wonder why their CPU goes 100%.

RC-Slater, 2016-03-08:
Check that your tx radio can drop your stick values to less than 1040.  That's the new default in beta. The old default is 1100.

borisbstyle, 2016-03-08:
yup make sure tx has a throw of 1000-2000 for best results :D

brauliolorenzo, 2016-03-09:
Hi. I have the same problem, but with spektrum satellite. 

When I use version 2.41 it´s work fine, but 2.53 or 2.52 now arm motors.

Can you help me?

Thanks, 
Bráulio Lorenzo

garrettben, 2016-03-09:
I have the same issue, also with spektrum satellite on a naze (beef's)
board.

garrettben, 2016-03-09:
Nevermind.  I just moved my throttle center to 1427 and my max and min
followed.  Seems the spektrum minimum with throttle center at 1500 is 1060.

Ben

borisbstyle, 2016-03-10:
I will close this issue. The manual always said to use 1000-2000 endpoints ;)

closed 2016-03-10 by borisbstyle