betaflight/betaflight #258: Betaflight could probably benefit from a rebase against Cleanflight

martinbudden, 2016-03-08:
Today github says "This branch is 739 commits ahead, 557 commits behind clean flight:master."

As divergence increases it will become progressively more complicated to pull code from Betaflight to Cleanflight and vice versa. A rebase will make that easier and also makes it easier to do differences between the two forks.

I know it is a bit of a pain doing a rebase, but I think it is beneficial. Now is a fairly good time to rebase, since there has just been a Cleanflight release, and there is a lull in the commits to Cleanflight.

(As a future recommendation, I would say it is a good idea to do a rebase every 20 or 30 commits to Cleanflight - when rebases are done frequently they are quite quick and easy to do.)

borisbstyle, 2016-03-08:
@martinbudden 
I know and I tried that, but it simply it wasn't possible back than. The reason was simply that rebase always introduced to many bugs for one person to handle.
It sometimes killed a lot of free time in the evenings troubleshooting those and especially the failsafe part in the past was responsible for the diversion. I believe that works now in the current cleanflight code base, but the diversion is too huge to do a full rebase without headache.
Still it is not that hard for me personally to port features once those are proven to work.

martinbudden, 2016-03-08:
It's a pity it's such a headache. Perhaps a rebase might be possible in future when Cleanflight has taken more Betaflight features. I guess you can close this issue, but I still think it's worth revisiting in future.

closed 2016-03-10 by borisbstyle