betaflight/betaflight #260: Dump command does not display all the Rate Profiles

TonyLeyland, 2016-03-09:
When issuing a Dump command the Profile section had the RateProfile included in it and also the Dump Rates section only displays the Rate Profile 0

Here is an example:

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 30
set tpa_rate = 0
set tpa_breakpoint = 1500

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 30
set tpa_rate = 0
set tpa_breakpoint = 1500

bevibes, 2016-03-09:
Hi,
Same for me and unable to use/configure/switch the third profile.

borisbstyle, 2016-03-10:
@bevibes 
There is no 3rd profile. It has been removed to make some room for new features.

@TonyLeyland 
That is all correct I believe. This was all again done to save flash space.

@KiteAnton
Is this all correct?

TonyLeyland, 2016-03-10:
I understand that there are only 2 Profiles (PIDs etc) to save space, but Betaflight 2.5.3 actually supports 3 **Rate Profiles**. This has always been a source of confusion to lots of people since the word **Profile** is used to describe the mechanism for encapsulating the PIDs etc, then the word **Rate Profile** is used to encapsulate just the Rates. What I have reported in this issue is correct as it was copied straight from Dump command.

I have a 3 position switch that selects one of the 3 different Rate Profiles. However, the Dump command has never been successful in displaying those rates, so I keep them in a separate text file and copy/paste them back into Betaflight every time I upgrade. See **example 1** below for what I paste back via the Command Line. I've used the ^ symbol to represent the hash symbol, since hash formats the text to a header ! 
You can see that it has the Profile and the 3 Rate Profiles. This is fully supported approach but the Dump command does not return the same as was entered.

I think the whole Profile & Rate Profile needs some re-work when time permits.
My suggestion would be as follows:

1) Come up with a new naming standard for Profile and Rate Profile that is not so confusing. Probably simply dropping the word Profile from the Rate Profile section.
2) Since we now have 2 Profiles to work with (due to space limitations), The Rates should be encapsulated within the Profile. Therefore, each of the 2 Profiles will have their own Rates.
3) Enable CF Configurator to have the ability to change Profiles via switch on the TX (like what can be done with Rate Profiles now). This will mean that when the switch is activated the GUI will show the new PIDs, PID Controller, Rates etc pertinent to the Profile that has been selected.

See **example 2** below for how the Dump command would look after these changes. I've used the ^ symbol to represent the hash symbol, since hash formats the text to a header !

The advantage of this approach is that you can change Profiles while flying and therefore have the ability to have different Rates based on the flying situation currently being faced (may be the wind has picked up and different Rate are going to help you). It also means that if the 2nd Profile happened to have a different PID values or even different PID Controller then you can make direct comparisions between PIDs or Controller while flying - no need to land and use Stick to change profile etc. 

Sorry for such a detailed report !

Cheers
Tony

## Example 1

^ dump profile

^ profile
profile 0

set yaw_p_limit = 500
set pid_delta_method = MEASUREMENT
set dterm_lpf_hz =  0.000
set dterm_average_count = 4
set pid_controller = MWREWRITE
set p_pitch = 65
set i_pitch = 40
set d_pitch = 26
set p_roll = 43
set i_roll = 40
set d_roll = 18
set p_yaw = 140
set i_yaw = 40
set d_yaw = 16
set p_pitchf =  1.400
set i_pitchf =  0.400
set d_pitchf =  0.010
set p_rollf =  1.100
set i_rollf =  0.400
set d_rollf =  0.010
set p_yawf =  4.000
set i_yawf =  0.400
set d_yawf =  0.000
set level_horizon =  4.000
set level_angle =  4.000
set sensitivity_horizon = 75
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 30
set i_level = 8
set d_level = 60
set p_vel = 120
set i_vel = 45
set d_vel = 1

^ rateprofile
rateprofile 0

set rc_rate = 100
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 30
set tpa_rate = 0
set tpa_breakpoint = 1500

rateprofile 1

set rc_rate = 100
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 40
set pitch_rate = 40
set yaw_rate = 50
set tpa_rate = 0
set tpa_breakpoint = 1500

rateprofile 2

set rc_rate = 100
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 60
set pitch_rate = 60
set yaw_rate = 70
set tpa_rate = 0
set tpa_breakpoint = 1500

## Example 2

^dump profile
^profile
profile 0
set pid_controller = MWREWRITE
set p_pitch = 65
set i_pitch = 40
set d_pitch = 26
.
.
etc

^rates for this profile
set rc_rate = 100
set rc_expo = 70
set rc_yaw_expo = 20
.
.
etc

^profile
profile 1
set pid_controller = MW23
set p_pitch = 42
set i_pitch = 27
set d_pitch = 14
.
.
etc

^rates for this profile
set rc_rate = 90
set rc_expo = 42
set rc_yaw_expo = 12
.
.
etc

mikeller, 2016-03-10:
Hi Tony.

All of this was discussed here a while ago.

To sum it up:

1.) Probably a good idea, the names are confusing.
2) This is already done. In the current implementation, each one of the profiles can have up to 3 independant rate profiles
3) Essentially, when profiles are switched, the FC sets the profile to load in permanent memory, and then does a reboot. Being able to do this in mid flight sounds like a terrible idea. ;-)

KiteAnton, 2016-03-12:
In dump, only rateprofile settings for current active rateprofile is outputted. This is similar as how only current profile settings is outputted when using normal dump. 

However, i made a quick test and it is only minor changes to output as in your example 1. I can make a PR for this soon. 

Should also all available profiles but outputted? And not only current as it is now.. ?

borisbstyle, 2016-03-12:
Thats a good idea Anton....make a pr

KiteAnton, 2016-03-12:
Ok created PR for this. I think this will make it easier for people to dump/restore settings when using multiple profiles and rateprofiles.

borisbstyle, 2016-03-12:
can you also test if  dump/restore works properly?

KiteAnton, 2016-03-12:
I tested this now and it works if not everything is pasted at the same time. 
When pasting in cli and the `profile x` is used it seems to loose some text from buffer, maybe it needs some additional pause for this. I guess this is the same also for normal dump also previous to this change. Maybe it also depends on the computer, I don't have a speed monster here so if someone else also can test the PR ( #273 ) with dump/restore then that would be great.

I am no expert in cli buffer/input so if someone knows how to hold the buffer/input slightly when switching profiles ?

Also when testing this I found there is a bug in the dump of beeper settings where it inserts too many spaces so pasting them back results in invalid value.

borisbstyle, 2016-03-12:
thats also the reason why it was never there together.
I remember something about that.

mikeller, 2016-03-12:
Yes, it was always the case that on restore, there had to be a pause after loading a new profile. I still think it is a good idea to have one command that dumps everything in one go, in order to make it easy to do a full back up.

As a suggestion, what about inserting a comment like

`# wait for one second or more here`

after every

`profile x`

command?

borisbstyle, 2016-03-12:
maybe we can introduce sleep in between?

So we dump like a sleep or pause command that will also be copied back to
make ot even easier.

mikeller, 2016-03-12:
Hmm, if anything, the sleep command would have to be executed by the cleanflight CLI, or whatever else is used to send the dump during the restore. As it is, the receiver (FC) loses the data that is sent while it is switching profiles, without the sender (cleanflight CLI / serial terminal) being aware of this.

KiteAnton, 2016-03-12:
Testing now, seems like just a couple extra blank lines will fix it but the problem is intermittent and it seems not to be only when switching profiles, but I guess it is most likely to show up there since that takes additional cpu load for the FC. 
I think for a long term solution the cli in the configurator needs to slow down output to allow FC to keep up.

borisbstyle, 2016-03-12:
it is indeed something what should go into configurator

KiteAnton, 2016-03-12:
I tried changing "waiting time" in configurator and that fixes the problem. Will make a PR also for the configurator.

borisbstyle, 2016-03-14:
fixed

closed 2016-03-14 by borisbstyle