betaflight/betaflight #261: Profile #3 unusable

Eganwp, 2016-03-09:
Hi Borris,

After several hours of testing last week, profile #3 appears to have a bug in that it is unselectable/unconfigurable ever since patch 2.4.1. It still doesn't work in 2.5.0 & 2.5.3 from my testing on the IRC F3 board. It did work fine last in 2.3.5.

The 3rd profile is there but if you try to select it, as soon as you change focus it jumps to profile #1 again. Ive confirmed this behaviour in both the latest version of CF as well as the EzOSD. Profile #3 can't be used or configured.

Thanks!
Travis

borisbstyle, 2016-03-10:
profile 3 has been removed in 2.4.1 :)

That had been done to free some room for new features. F1 targets were
getting full

Eganwp, 2016-03-10:
Oh! That makes sense. I was confused because I thought one of the release candidates for 2.5.0 said it was added back again, but obviously I must have misread it.

Thanks Borris! Great job on all your hard work btw. I really appreciate it and always get excited every time I refresh the page and a new FW revision is out. ;)

Travis

closed 2016-03-10 by borisbstyle