betaflight/betaflight #263: Beeper tone on startup can't be deactivated (repost)

lblblb, 2016-03-10:
Hi,
I created issue 218: https://github.com/borisbstyle/betaflight/issues/218
and it got closed because the beeper commands got reworked and the initialisation beeps should be possible to get deactivated. I'm now running BF 2.5.3 with the new commands, configured everything and the dump is:

`# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper  RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB`

But the beeper **still does the initialisation beeps** after plugging the LiPo in. When the board (Lumenier Lux) gets powered on by USB this does not happen, however.

lblblb, 2016-03-10:
I don't even have any motors connected at the moment. It really is the beeper.

Flugsicherung, 2016-03-11:
I have the same Problem, but I found my issue. I have changed the min voltage per cell to 4.2 and this was the Problem. If I connect a full battery the FC, hi thing this is a 4S battery and so the voltage is too low. This is the beeping!
I change the setting to 4.3 and everything is OK.

@borisbstyle 
Is it posible to show how many cells the FC find when hi starts in the configurator start tab?

borisbstyle, 2016-03-11:
You  need to have 4.3 than otherwise a full battery of 4,21 or 4.22 may be
detected as more cells. The very small deviation on the voltage makes ot
also easier to get above 4.2 especially after startup.

lblblb, 2016-03-11:
So I need to leave the settings as they are? I don't even have VBAT / voltage monitoring enabled, this should not cause the beeping regardless of the voltage settings, as it's not active.

borisbstyle, 2016-03-11:
feature VBAT is enabled by default

lblblb, 2016-03-11:
I disabled it long ago and still the beeper does the intitialisation beeps:
[image]
I also disabled it via CLI for good measure but it doesn't help at all.

borisbstyle, 2016-03-11:
than disable beeper_off_flags for vbat too

lblblb, 2016-03-11:
It already is. My first post shows the CLI dump (section beeper) of my setup. Are there any more settings than these?

borisbstyle, 2016-03-11:
your rx set is still enabled

lblblb, 2016-03-13:
RX_SET does not affect my problem. See, I now set it to beeper -ALL with this dump:

beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -MULTI_BEEPS
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB

saved it, disconnected from computer and plugged in LiPo. Beeper tones still there on startup.

robertlacroix, 2016-03-13:
Check if your BEEPER mode is enabled at 1500. If it is change it to something higher.

lblblb, 2016-03-13:
My BEEPER mode gets activated above 1700 µs via a radio switch, but this switch activated beeper is beeping with a lower frequency than the unwanted initialisation beeps are. These initialisations beeps in in very fast succession.

nljackson, 2016-04-05:
Howdy, I'd like to report that I am experiencing the same issue. beeper -ALL and I get a rapid beep when plugging in the lipo with no usb. If I plug in the usb first, then plug in the lipo I get no beeps which is the desired behavior.

lblblb, 2016-04-28:
I've looked on my own for the initialisation beeping sequence, and it's in main.c starting at line 477:

    for (i = 0; i < 10; i++) {
        LED1_TOGGLE;
        LED0_TOGGLE;
        delay(25);
        if (!(getPreferedBeeperOffMask() & (1 << (BEEPER_SYSTEM_INIT - 1)))) BEEP_ON;
        delay(25);
        BEEP_OFF;
    }

Could anyone confirm that the relevant if statement is set up correctly? Again, the issue is that these beeps occur even if beeper_system_init is disabled.

tsly, 2016-05-18:
I have the same issue. On top of it there is one more annoying 'feature' as well: When I'm using a beeper switch on a spare tx channel it is working only when RX_SET beeper flag is active. I think it would be more elegant to use a separate beeper flag for using the beeper when I flip the switch.

KiteAnton, 2016-05-25:
I checked the code and I think I found the problem. 
This line, https://github.com/borisbstyle/betaflight/blob/betaflight/src/main/main.c#L483, 
Should use getBeeperOffMask() instead of getPreferedBeeperOffMask(). 

The getPreferedBeeperOffMask() uses a parameter(masterConfig.prefered_beeper_off_flags) that seems to be unused (and never set). That function and parameter can be removed as well.

If someone wants to test this (I have not yet) and make a PR then do it. Otherwise I will perhaps be able to do it in a couple of days.

mikeller, 2016-05-25:
I've annoyed everyone in the office with the initialisation beeps often enough, I'll look into it. Good spotting, @KiteAnton.

mikeller, 2016-05-25:
@KiteAnton: I've fixed the beep on init bug. As to the `setPreferedBeeperOffMask()` / `getPreferedBeeperOffMask()`, this seems to be used to store / retrieve an alternative set of beeper settings, and is used in `serial_cli.c`. `beeper preferred` stores the currently active beeper settings to the alternative location, `beeper -preferred` restores them from the alternative location. @borisbstyle has added this code, so he can probably explain what use case this is for.

PR #421 (2016-05-25): Fixed failure to disable beeping on FC init. Also, fixed spelling.
Fixes #263.
closed 2016-05-25 by borisbstyle