betaflight/betaflight #265: new to betaflight does latest version have pin 6 buzzer

damovdv, 2016-03-10:
Hi,  

I am new to betaflight and was wondering does this latest version still have pin 6 buzzer support  for cc3d board.

Thanks

D

GareFPV, 2016-03-13:
I am on Betaflight 2.5.3 too and my CC3D Buzzer on Pin 6 didnt work:

set enable_buzzer_p6 = ON
beeper -ALL
beeper RX_SET

borisbstyle, 2016-03-14:
@GareFPV 
Whats the last version where it did work?

GareFPV, 2016-03-17:
I have found out randomly, that the problem is my CC3D. I have tested the previously Betaflight version 2.4.1 with this CLI-commands:

set enable_buzzer_p6 = ON
set beeper_off_flags = 65023

Nothing :-( I buy some new buzzer and they didnt work too. Sorry for my mistake-post. My Hobbyking-buzzer work fine one week ago...

closed 2016-03-18 by borisbstyle